ros-industrial-attic/ur_modern_driver #5: Feature request: Lock teach pendant when under ros_control

ThomasTimm, 2015-10-13:
With the dashboard server, it is possible to lock the teach pendant from firmware version 3.1.17something by setting the user role to 'restricted' (http://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/dashboard-server-port-29999-15690/)
This can prevent the user from trying to use the screen and accidently crash the controller.
The enhancement should also display a message on the teach pendant to let them know that they are locked out.
The enhancement could be extended to be active whenever the driver is executing something on the robot (whether under ros_control or action interface). This would probably improve the stability of the driver.
Nevertheless it is quite important that the driver always unlocks the screen when the driver shuts down! Perhaps opening a new connection to the dashboard server, unlocking everything when the driver is killed, is the best way to go.

As I haven't tried it out, I am uncertain on how this works. The documentation says that it prevents the user from pressing the 'move' tab. If this is indeed the case and it is thus not possible to read the joint states and end effector position from the screen, then this enhancement should be made dependent on a parameter in the driver, allowing the user to turn it off.

hemes, 2016-01-12:
I tried this out and after sending the command "setUserRole restricted" the pendant returned to welcome screen and the move tab was no longer present on subsequent menus.

gavanderhoorn, 2019-11-23:
`ur_robot_driver` adds *External Control* urscript integration which clearly shows *when* the robot is being controlled externally.

There is also Dashboard server integration available, with a ROS API.

closed 2019-11-23 by gavanderhoorn