ros-industrial-attic/ur_modern_driver #18: Transfer ur_modern_driver to ros-industrial/universal_robots?

ipa-fxm, 2015-11-14:
As discussed with @ThomasTimm and @shaun-edwards and during ROS-I Developer Online Meeting, transfering this new driver to the ROS-I repo (https://github.com/ros-industrial/universal_robot) would be a great thing.

This issue is for discussion on how this can be done in the smoothest way.

@ThomasTimm 
Is there any current issue or PR (I don't see any) that blocks the transfer, i.e. should be resolved prior to the transfer?

This [tutorial](http://stuermer-benjamin.de/programming/moving-files-slash-directories-between-git-repositories-without-loosing-history/) shows how a repo can be moved between repos without loosing history.
Is there also a way to transfer issues to the new repo? I've seen thing like that, but I don't know how this can be done...

ThomasTimm, 2015-11-16:
No, I don't have any current issues. It would be great to have some documentation up describing how to use the driver and advising how to transition from the old driver to this, but code-wise there are no problems.

I don't have that much experience with git and Github, so I have no idea, but it would be great if the issues could be transfer. If not, it shouldn't be to difficult to type them in again, there are hardly, if any, issues that has any comments.

shaun-edwards, 2015-11-16:
I'm not aware of a way to transfer issues.  One solution is to make a single issue in the ROS-I repo with a checklist that lists the open issues in @ThomasTimm's repo.

ipa-fxm, 2015-11-19:
For moving issues, have a look at https://github.com/google/github-issue-mover!
Also, [this discussion](https://github.com/holman/ama/issues/413) might be relevant...

ThomasTimm, 2015-11-20:
None of the existing issues have any comments, and I have mostly written them to remind my future self of changes I would like to make. My time is quite limited for the rest of 2015, so it is not likely that much will be fixed before jan/2016. But if it is that easy to move the current issues, I think we should definitely do it. It also serves as a nice roadmap for users to see where this package is going.

I looked at the ros-industrial/universal_robot issues, and from what I can tell, we should be able to close ros-industrial/universal_robot#216, ros-industrial/universal_robot#201, ros-industrial/universal_robot#199, ros-industrial/universal_robot#191, ros-industrial/universal_robot#188, ros-industrial/universal_robot#183, ros-industrial/universal_robot#178, ros-industrial/universal_robot#177, ros-industrial/universal_robot#176, ros-industrial/universal_robot#118, ros-industrial/universal_robot#80, ros-industrial/universal_robot#59, ros-industrial/universal_robot#54 and ros-industrial/universal_robot#28 with this merge. ros-industrial/universal_robot#207 still exists as I am using the same interpolation method, but maybe using ros_control adheres to these limitations?

shaun-edwards, 2015-12-02:
Is anything holding us back from transferring this?  There is still one open PR.

ThomasTimm, 2015-12-02:
The PR is for creating a non-ros interface. It is thus not relevant for the merger. I think it would make more sense to cancel that PR, merge this repo into the universal-robot and then branch that off into a separate repo for non-ros users.

The two things holding me back is not having access to merge anything into the universal-robots repo and lack of time. Sadly I won't be able to put much work into this in the current month.

hecatezxx, 2015-12-17:
I have some questions about this driver. Is there any solution to control the velocity under cartesian space? I have found that controlling velocities requires joint position data. Is it means that i should transform the cartesian path data to joint position data?
Additionally, could i use this driver with ROS-Indigo? I saw it claims that it cannot directly be used with ROS versions prior to hydro.

ThomasTimm, 2015-12-17:
@hecatezxx : As your question has nothing to do with the issue in which you have posted, would you mind re-posting it on the ros-industrial mailing ( https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/qo9pu4PbEJY )

hemes, 2016-01-13:
One issue I just found when making some modifications is that the code doesn't adhere to the [ROS C++ style guidelines](http://wiki.ros.org/CppStyleGuide) .  I don't know how picky the ROS-I guys are but just wanted to bring it up.  In particular the formatting has the following issues:
- Literal tab characters instead of spaces
- Curly braces not on their own lines
- Line length? (didn't actually count)

shaun-edwards, 2016-01-15:
We are pretty picky when it comes to style.  Luckily the issues you identified can easily be fixed with a IDE like QtCreator or Eclipse (with the right styles loaded).  I recommend making these types of style changes as a single PR with no other changes.

vickailiu, 2016-09-08:
Hi @ThomasTimm , I'm a bit lost with it. May I know which I shall follow if I want to start a new project with the UR robot, from your ur_modern_driver or https://github.com/ros-industrial/universal_robot. Kindly Advise.

Also, I'd like to implement my UR control logics by C++ instead of python, may I know if there is any existing samples that I can follow just like the test_move.py?

BrettHemes, 2017-01-31:
@ThomasTimm @shaun-edwards any progress on this front?  I recently made (mostly) the switch to Kinetic and this driver is lacking support.  #58 has been open for some time but I figure these could both be accomplished at once when moved.

gavanderhoorn, 2018-10-05:
Closing this one as it's here (on `ros-industrial` that is).

I'll go through the issue listed by @ThomasTimm in https://github.com/ros-industrial/ur_modern_driver/issues/18#issuecomment-158370977 and close those that can be closed but haven't yet been.

closed 2018-10-05 by gavanderhoorn