ros-industrial-attic/ur_modern_driver #25: Corrupted joint_states are published with ur_sim and real robot (1.8.x).

jettan, 2015-12-01:
While using the driver to interface with both (but not at the same time) a ur-sim instance (v1.8.14035) and a real controller (CB2, also v1.8.14035), we sometimes observe `JointState` messages that have some really strange values in some of the `position` and `velocity` arrays. An example is pasted below:
```yaml
header: 
  seq: 17396
  stamp: 
    secs: 1448973318
    nsecs: 95394434
  frame_id: ''
name: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
position: [0.0008817444019501054, 0.0, 0.0, 0.0, -4.737910152973887e+226, 3.4403246606906987e-246]
velocity: [5.356556342585614e-69, -4.555528633092074e-39, -3.501131509392264e-164, -4.0975983370358134e-95, 2.139376420684511e+253, 1.1701194070372343e-307]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 8.0963e-320]
```
Notice the `-4.737910152973887e+226` and `3.4403246606906987e-246` in the `position` array.

Could this be an off-by-one error in the de-serialisation routines?

jettan, 2015-12-01:
We should probably provide you with some wireshark/tcpdump traces to help debug this?

ThomasTimm, 2015-12-01:
Yes, a wireshark file or tcpdump would help a lot :)

I think you are right in that it is an off-by-one error, but I don't see why. So if you can supply a datadump, preferably also with the previous ~10 data packages before the one that gives odd results, that would be a great help.

jettan, 2015-12-02:
Since I cannot upload a pcap or compressed archive to github, please find it in the dropbox link below:
https://www.dropbox.com/s/6gados5nn7g5eq5/attachments.tar.bz2?dl=0
Also included is the log file as requested of the rostopic /joint_states.

ThomasTimm, 2015-12-02:
Thanks, I'll look into it. Unfortunately my time to work on the project is very limited the next month, so it might be a while until I have it fixed.
If anybody else has the time to look at it, pull requests will be much appreciated.

jettan, 2015-12-02:
I'll look into it myself too and see if I can come up with a pull request.

closed 2015-12-02 by ThomasTimm

ThomasTimm, 2015-12-03:
FIY I updated the code, so now it does the check for all current firmware versions

PR #26 (2015-12-02): Ignore malformed messages from UR5 running 1.8.X.
Fixes #25 
Here, we check if the length of the message is actually 812 bytes long and ignore them if they are not. Every 17th and 18th message is 560 bytes followed by 9 bytes and were interpreted anyway causing the driver to publish corrupted joint states.