ros-industrial-attic/ur_modern_driver #27: ur5_ros_control totally unstable

ibaranov-cp, 2015-12-14:
Hello all,

Perhaps I am doing something wrong..
With a UR5 plugged in and working normally (Driver 3.2.18654), I execute:

roslaunch ur_modern_driver ur5_ros_control.launch robot_ip:=192.168.1.17

The robot then displays an unstable increasing oscillation in a random direction until I am forced to kill it.

Has the controller been tested on real robots?

ThomasTimm, 2015-12-15:
Yes, I have tested it on real robots, although the UR5 and UR10 has only been with firmware 1.8.
And most of my testing have been on the UR10, so the controller gains might not work very well for the UR5.
Try to lower the gains in the config/ur5_controllers.yaml file; that should help on the instability.
Some time early next year I will hopefully finish a tutorial on getting ros_control to work, tuning the controllers, and which topics to look at for troubleshooting.

Please also note that I haven't modified the driver to work with firmware 3.2 yet. The new firmware introduces a few changes to the communication protocols, but nothing that explains what you are experiencing.

cpacchierotti, 2016-01-19:
Dear all,

I am experiencing the same issue (unstable increasing oscillation in a random direction) with a UR5 and controller 3.2.

If I run

terminal 1: ```roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=ROBOT_IP```, and
terminal 2:  ```rosrun ur_driver test_move.py```

The robot does **not** move, and the outputs are

terminal 1
```
Wrong length of message on RT interface: 1072173060
Wrong length of message on RT interface: 1072173060
Wrong length of message on RT interface: 1072022871
[ INFO] [1453194748.736263756]: on_goal
[ERROR] [1453194748.736418567]: Goal start doesn't match current pose
```

terminal 2
```
Waiting for server...
Connected to server
```

Any suggestion?

Thank you very much.

ThomasTimm, 2016-01-19:
As I suggested to Ibaranov; have you tried lowering the gains and see if that helps with the instability?  -I'm eager to know whether that solved it (i.e. the gains are just a bit to high)

As mentioned in Issue #17 version 3.2 is not yet fully supported. The first three warnings on terminal 1 are a bit strange though, they suggest that you received some very long messages when first connecting. But fortunately it seems to only concern the three first messages, and doesn't prevent the driver from working.

The error related to the fact that the test_move.py from the original driver doesn't make the robot move, is related to the test_move.py script. That script just assumes that the robot is at some position. As your robot is not at the assumed position, the driver (correctly) discards the trajectory for safety reasons. 
There are three ways to fix this:
1. Move your robot to the pose specified in move_test.py (easiest, but not very robust)
2. Change the starting pose in move_test.py to match your robot. (almost as easy, just as fragile)
3. Change move_test.py to subscribe to joint_states and get the current robot pose from there (hardest, but most robust solution. And, once you done it, submit a pull request to the original ur_driver to fix the bug)

cpacchierotti, 2016-01-19:
Dear Thomas,

with a P value of 1.2 it does not oscillate. Higher values cause oscillations.

However, I am not able to test more the PID values since I still can't find an easy way to remotely move the robot around. Can you point me to an sample C++ code to move the robot to predefined target positions? I have an UR5 with controller 3.2.19293 (the latest) and I would like to test the new drivers a little bit more.

Thank you.

On a side note, the message
```Wrong length of message on RT interface: 1072173060```

comes very often (5-10 times per minute ). I am opening another thread to avoid going off topic.

ThomasTimm, 2016-01-19:
Thanks for trying it out, I am glad that it was just the controller values that needed tweaking.

The easiest way to start moving the robot around is by copying the test_move.py script from the ur_driver package to your workspace, modify it and change the values for Q1 on line 11 to match the current pose of your robot.
To get the current values in a manner where you can just copy/paste, use 'rostopic echo /joint_states' in a terminal with the driver running.

Please note the error message in issue #17 then I'll have a look at it when time permits.

closed 2016-02-19 by ThomasTimm

Chunting, 2016-05-23:
Hi,
When I run the same command as you, 
roslaunch ur_modern_driver ur5_ros_control.launch robot_ip:=192.168.1.102

It failed with the information below.

```
NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ros_control_controller_manager (controller_manager/controller_manager)
    ros_control_controller_spawner (controller_manager/spawner)
    ur_hardware_interface (ur_modern_driver/ur_driver)

WARNING: WARN: unrecognized 'arg' tag in <node> tag. Node xml is <node launch-prefix="$(arg launch_prefix)" name="ur_hardware_interface" output="screen" pkg="ur_modern_driver" type="ur_driver">
    <param name="robot_ip_address" type="str" value="$(arg robot_ip)"/>
    <param name="min_payload" type="double" value="$(arg min_payload)"/>
    <param name="max_payload" type="double" value="$(arg max_payload)"/>
    <param name="max_velocity" type="double" value="$(arg max_velocity)"/>
    <param name="use_ros_control" type="bool" value="True"/>
    <arg name="prefix" value="$(arg prefix)"/>
  </node>
ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[ur_hardware_interface-1]: started with pid [19117]
process[ros_control_controller_spawner-2]: started with pid [19118]
process[ros_control_controller_manager-3]: started with pid [19119]
process[robot_state_publisher-4]: started with pid [19126]
[ INFO] [1464007966.784880219]: Reading rosparams from namespace: /
[ INFO] [1464007966.785639185]: Loaded ur_hardware_interface.
[FATAL] [1464007966.831960481]: Error connecting to get firmware version
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[INFO] [WallTime: 1464007967.144302] Controller Spawner: Waiting for service controller_manager/load_controller
[ur_hardware_interface-1] process has died [pid 19117, exit code -6, cmd /home/chunting/catkin_ws/devel/lib/ur_modern_driver/ur_driver __name:=ur_hardware_interface __log:=/home/chunting/.ros/log/d1d1ecba-20de-11e6-99d4-44850011896a/ur_hardware_interface-1.log].
log file: /home/chunting/.ros/log/d1d1ecba-20de-11e6-99d4-44850011896a/ur_hardware_interface-1*.log
[WARN] [WallTime: 1464007997.316460] Controller Spawner couldn't find the expected controller_manager ROS interface.
[ros_control_controller_spawner-2] process has finished cleanly
log file: /home/chunting/.ros/log/d1d1ecba-20de-11e6-99d4-44850011896a/ros_control_controller_spawner-2*.log
```
When I run the command below in another terminal,
rosrun ur_driver test_move.py

It shows a message " Waiting for server"
I have no idea what happens, would you please give me some suggestions?

Thanks.
Chunting

Chunting, 2016-05-23:
My controller is CB 3.2.18744, ros is indigo

commit c05f16c8 (2016-02-19): Tweaked ur5 PID controller values. Fixes #27