ros-industrial-attic/ur_modern_driver #30: Check for return value of socket_open in URScript

ThomasTimm, 2016-01-05:
When using the driver with a simulated robot on a virtual box with poorly configured network, the driver might be able to see the robot and attempt a connect, but the box might not be able to connect to the driver with URScript.

A suggested solution is to check the return value of socket_open in the URScript and alert the user with a popup, if the reverse connection is unsuccessful.

This is seldom a problem (where the driver can connect to the robot and upload URScript, but the robot cannot open a reverse connection), and should be compared to the added amount of data to be transferred to the robot on each new trajectory. Thus this issue will probably remain open for some time to allow better solution suggestions.

gavanderhoorn, 2016-01-06:
> When using the driver with a simulated robot on a virtual box with poorly configured network, the driver might be able to see the robot and attempt a connect, but the box might not be able to connect to the driver with URScript.

I can imagine a (misconfigured) firewall on the PC running the ros node could result in the same issue.

Any time the PC connects to the robot through some network interface that does not (automatically) allow return traffic (over a different port), such as a NAT bridge, a firewalled connection or something else, the same issue will arise.

gavanderhoorn, 2016-01-06:
Also: the described behaviour is actually more like a bug. Ignoring return values should always be the exception, especially with network programming, where so much can go wrong.

ThomasTimm, 2016-01-06:
I agree that optimally the driver would check for this, and I am happily accepting pull request til resolve the matter :)
But as I have ever only heard from 1 person where this has been an issue, and seeing that such a feature has never been requested for the old driver, it will not get top priority from me at the moment.

Hemes, who originally posted this problem on the mailing list, later suggested that the controller could be instructed to load the script via the dashboard server. If this does not introduce any significant delay I think it is the best solution. Thus, once I get around to implement an dashboard server interface, I will test that solution as well.
This would require the user to install the URScript on the controller, but that is not different from most other ROS-Industrial driver. Alternatively the driver package could upload the script directly to the controller via FTP.

gavanderhoorn, 2016-01-07:
> Hemes, who originally posted this problem on the mailing list, later suggested that the controller could be instructed to load the script via the dashboard server. [..] This would require the user to install the URScript on the controller, but that is not different from most other ROS-Industrial driver.

A lot of users I talk to actually consider the fact that the ur driver(s) don't need to be installed one of the 'charms' of it (them). It would be a shame if that were to be removed.

As to the dashboard server: the reason being that it becomes possible to alert the user? Or to have a more visible running status on the TP? Displaying messages to the user can always be done, that should not need the dashboard.

> Alternatively the driver package could upload the script directly to the controller via FTP.

Is this a recent addition? I do not remember there being an FTP server on any UR controller I've ever worked with. Can also not find it in any of the manuals.

ThomasTimm, 2016-01-07:
The idea is to save the URScript on the controller, and the use the dashboard server to load that script. That way, the script wouldn't have to be transferred to the controller each time a new trajectory is executed. Whether there is anything to gain with this approach is yet to be seen though.

The controller doesn't have an ftp server, but the computer is running Linux, so it should be possible to copy the script via ftp (or perhaps rather via scp)
But I agree with you and the many users that it very convenient that it is not necessary to install anything on the controller. I'll see what is possible when time permits.

hemes, 2016-01-11:
The controller does have and FTP server that is always running (I actually use this quite frequently).  Per the CB3_Advanced_Training PowerPoint slides:
 - Server on UR controller
 - Server is always running on UR controller
 - No settings on UR controller needed
 - Details
    - Host: robot_ip_address
    - User: root
    - Password: easybot
    - Port: 22

If the password is never changed, the driver could transfer the file when initially invoked (using a single-line ftp operation) and then invoke the program via the dashboard server.

gavanderhoorn, 2016-01-11:
So it is a 'recent' addition.

Afaik CB2 controllers don't have that, so either it'll have to be optional or this driver will not be compatible with those controllers anymore then.

gavanderhoorn, 2016-01-11:
@hemes: what advantage do you see compared with just uploading the program every time?

hemes, 2016-01-11:
@gavanderhoorn: none really other than the transfer overhead.  Thomas' reservation to my original suggestion of adding a success/failure return value was due to increasing the size of the transferred script by ~10% and thus increasing the lag between when the trajectory is requested and when it actually executes.  By loading the file via the dashboard at least this time would be bounded?

dpsolomon, 2016-03-31:
I've had problems connecting to the reverse port - it was in fact a firewall issue.  The controller logs contain a note to the effect that it won't connect, but if you don't know to look in the logs, it's quite annoying to debug.

closed 2019-11-23 by gavanderhoorn

gavanderhoorn, 2019-11-23:
The *External Control* urcap clearly indicates when it cannot connect to the remote (ROS) PC with a regular popup on the TP.

This should avoid the issue described in the OP.