ros-industrial-attic/ur_modern_driver #33: [question] will the old ur_driver/test_move.py work with the new driver without modification?

MetaMemoryT, 2016-02-06:
Just wondering if [test_move.py](https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_driver/test_move.py) has been ran with the new driver.

MetaMemoryT, 2016-02-06:
I think I found the answer to my question here:

https://github.com/ThomasTimm/ur_modern_driver/issues/27

``
The easiest way to start moving the robot around is by copying the test_move.py script from the ur_driver package to your workspace, modify it and change the values for Q1 on line 11 to match the current pose of your robot.
``

closed 2016-02-06 by MetaMemoryT

ThomasTimm, 2016-02-19:
I have just added a modified version of test_move.py to this repository, which subscribes to the joint_states topic and thus gets the correct starting position. 
For future users, please use this file