ros-industrial-attic/ur_modern_driver #37: crash on using test move

ibaranov-cp, 2016-03-10:
Hello,

using ur5_bringup.launch looks ok, however when test_move.py is run, after hitting y to continue, the driver crashes with this result:

[roslaunch][INFO] 2016-03-09 19:36:45,064: process[ur_driver-2]: starting os process
[roslaunch][INFO] 2016-03-09 19:36:45,064: process[ur_driver-2]: start w/ args [[u'/home/administrator/catkin_ws$
[roslaunch][INFO] 2016-03-09 19:36:45,064: process[ur_driver-2]: cwd will be [/home/administrator/.ros]
[roslaunch][INFO] 2016-03-09 19:36:45,065: process[ur_driver-2]: started with pid [3936]
[roslaunch][INFO] 2016-03-09 19:36:45,065: ... successfully launched [ur_driver-2]
[roslaunch][INFO] 2016-03-09 19:36:45,065: ... launch_nodes complete
[roslaunch.pmon][INFO] 2016-03-09 19:36:45,066: registrations completed <ProcessMonitor(ProcessMonitor-1, starte$
[roslaunch.parent][INFO] 2016-03-09 19:36:45,066: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2016-03-09 19:36:45,066: spin
[roslaunch][ERROR] 2016-03-09 19:36:49,633: [ur_driver-2] process has died [pid 3936, exit code -11, cmd /home/a$
log file: /home/administrator/.ros/log/e9b182f4-e656-11e5-82ae-bd59d694d84e/ur_driver-2*.log
[roslaunch.pmon][INFO] 2016-03-09 19:36:49,633: ProcessMonitor.unregister[ur_driver-2] starting
[roslaunch.pmon][INFO] 2016-03-09 19:36:49,633: ProcessMonitor.unregister[ur_driver-2] complete

ThomasTimm, 2016-03-10:
What is written in the log file: /home/administrator/.ros/log/e9b182f4-e656-11e5-82ae-bd59d694d84e/ur_driver-2*.log ?

ibaranov-cp, 2016-03-10:
Unfortunately, it looks like that log is never actually written, file does
not exist... Perhaps a crash prior to diagnostic output. Will keep digging

ThomasTimm, 2016-04-12:
I have had the issue of no log being written as well.
I don't know if I am missing something in my code, or if the missing log is due to the fact that the process dies unexpectedly.
Anybody knows a good explaination?

ThomasTimm, 2016-05-31:
I think I've found the answer to my own question; try changing the launch file to output to log instead of screen.

ThomasTimm, 2016-10-24:
Any news on this, or can we close this issue?

gavanderhoorn, 2017-11-24:
I'm going to close this one as we need more info to diagnose and reproduce this.

If this is still a problem @ibaranov-cp, then please update and re-open.

closed 2017-11-24 by gavanderhoorn