ros-industrial-attic/ur_modern_driver #39: Feature request: rate limit publishing of topics

ibaranov-cp, 2016-03-13:
Hello,

Most of the topics this driver publishes are fairly fast (100hz or more). In most cases this is great, however if you are working on a bandwidth limited system, this quickly swamps transfer.

Setting a param for publishing rate would be great, not high priority.

ThomasTimm, 2016-04-12:
I think all the data is published at 125 Hz, which is the update frequency of the controller.
It would be fairly easy to implement a divider to only publish every 2nd, 3rd etc... status message.
The publishing fre

ThomasTimm, 2016-04-12:
I think all the data is published at 125 Hz, which is the update frequency of the controller.
It would be fairly easy to implement a divider to only publish every 2nd, 3rd etc... status message.
The publishing frequency would then not be an integer number, but I don't see that is an issue.

Is that a feasable solution?

xqms, 2017-03-24:
Note that you can also simply use the `throttle` node from `topic_tools` package to apply external throttling to arbitrary rates without changing anything in the driver.

gavanderhoorn, 2019-06-29:
I believe the suggestion by @xqms makes sense, as it works in all cases.

The `ros_control` side of the driver actually supports configurable publication rates, so that would seem to be a viable option if using `throttle` is not possible/desirable.

closed 2019-06-29 by gavanderhoorn