ros-industrial-attic/ur_modern_driver #44: ros_control update timing

nikhilkalige, 2016-04-25:
I used this project as a reference when I was implementing ros_control driver for my robotis arms. I found that I was unable to send a trajectory to the arm if I made use of the following code for the header of the trajectory 
```python
self._goal.trajectory.header.stamp = rospy.Time.now()
```
But the following worked, as this would set the header.stamp to zeros
```python
self._goal.trajectory.header.stamp = rospy.Time(0)
```

I think this is because of the following lines in the `ur_ros_wrapper.cpp` where the `controller->update` is called, I think changing it to this would make it work. I haven't tried this out with a ur5, this is just my opinion based on using it in my code
```cpp
controller_manager_->update(
					ros::Time(current_time.tv_sec, current_time.tv_nsec),
					elapsed_time);
// Changed to
controller_manager_->update(ros::Time::now(), elapsed_time);
```

Took me 3 days to debug this.. :(

ThomasTimm, 2016-04-26:
I've been looking a bit into this, as I at first didn't understand what the issue is. I haven't had any problems using ros_control, neither have I heard of anybody else.

Apparently, the current_time.tv_nsec is a long, while the ros::Time contructor expects a uint32_t. This could give problems on some platforms. So it should definitely be modified.

I still believe the same time point should be used throughout the loop, although I recognize that Dave Coleman uses the same approach in his ros_control boilerplate that you suggest. I think I'll try to get his thoughts on the matter and see if we can agree on something.

ThomasTimm, 2016-04-29:
I'm still pondering the cause of your problem, as I don't really have enough information to recreate the issue - like I've said, I haven't heard of it before.

Could you try with 
`controller_manager_->update(
                    ros::Time((uint32_t)current_time_.tv_sec, (uint32_t)current_time_.tv_nsec),
                    elapsed_time);`
and see if that fixes the issue?

nikhilkalige, 2016-04-29:
I can try that out and get back to you on that this Saturday. With regards to my config, I am using Ubuntu 15.10 with jade, but all of it is compiled by myself as there are no existing ppa's. I am not sure if this is causing a problem, but I will try to get more logs with regards to that.

ThomasTimm, 2016-04-29:
I was especially curious as to the size of a long on your system. If your system makes a long as 64 bit little endian, the implicit typecasting might be the culprint - although I'm admittedly blindly guessing on that.
Would really appreciate the feedback, thanks for helping with figuring it out.

nikhilkalige, 2016-04-29:
I used the below code
```
std::numeric_limits<long>::max()

Value = 9223372036854775807
```
So its a 64 bit system and also its little endian

miguelprada, 2016-05-02:
I've just faced the same issue myself, and I think the problem comes from using `CLOCK_MONOTONIC` for filling out the `current_time`.

When the trajectory command comes filled with a timestamp, this is interpreted as the desired time for the trajectory to start by the joint trajectory controller in `ros_controllers`. If this timestamp is filled in using `ros::Time::now()`, then it will be the time from the Epoch, and in my experience, the time from the `CLOCK_MONOTONIC` is **much** smaller than that. This causes the trajectory to be accepted, but scheduled for very long into the future.

I cannot think of any issues with the approach proposed by @nikhilkalige, and in addition this is exactly how it is done in the [ros_control_boilerplate](https://github.com/davetcoleman/ros_control_boilerplate/blob/jade-devel/src/generic_hw_control_loop.cpp).

ThomasTimm, 2016-05-03:
Ahh yea, I remember seeing that in the discussion about defining ros::Time::now ( https://github.com/ros/roscpp_core/issues/33 ), but didn't take note at that time. Obviously, that is the problem. I haven't had any issues with it myself though (not that I've have noticed at least).

The reason I'm hesitating on the approach suggested by @nikhilkalige is that we acquire two different time points, with a non-deterministic read in between. See the issue https://github.com/davetcoleman/ros_control_boilerplate/issues/4 in the ros_control_boilerplate. I suggest we keep this discussion in that thread until we agree on the best solution to this.

miguelprada, 2016-05-03:
Ok, I'll keep an eye on that thread too. Thanks for the quick response!

nikhilkalige, 2016-05-10:
@ThomasTimm  I tried the code snippet you showed above `ros::Time(....)`, but that doesn't work too.

The log from the trajectory controller using `ros::Time`
```
ros.manipulator_driver.actionlib: Setting the current goal as succeeded
ros.manipulator_driver.actionlib: Setting status to succeeded on goal, id: /basketball_robot-2-1462838648.727, stamp: 1462838648.73
ros.manipulator_driver.actionlib: Publishing result for goal with id: /basketball_robot-2-1462838648.727 and stamp: 1462838648.73
ros.joint_trajectory_controller.actionlib: The action server has received a new goal request
ros.joint_trajectory_controller.joint_trajectory_controller: Recieved new action goal
ros.joint_trajectory_controller: Figuring out new trajectory starting at time 1462838651.701
ros.joint_trajectory_controller: Using alternate time base. In it, the new trajectory starts at time 1462814182.601
ros.joint_trajectory_controller: Trajectory has 3 segments:
- 1 segment(s) will still be executed from previous trajectory.
- 1 segment added for transitioning between the current trajectory and first point of the input message.
- 1 new segments (2 points) taken from the input trajectory.
ros.joint_trajectory_controller.actionlib: Accepting goal, id: /basketball_robot-3-1462838649.701, stamp: 1462838649.70
```

Logs when `ros::Time::now()` is used
```
ros.manipulator_driver.actionlib: Setting status to succeeded on goal, id: /basketball_robot-2-1462838723.998, stamp: 1462838724.00
ros.manipulator_driver.actionlib: Publishing result for goal with id: /basketball_robot-2-1462838723.998 and stamp: 1462838724.00
ros.joint_trajectory_controller.actionlib: The action server has received a new goal request
ros.joint_trajectory_controller.joint_trajectory_controller: Recieved new action goal
ros.joint_trajectory_controller: Figuring out new trajectory starting at time 1462838729.239
ros.joint_trajectory_controller: Using alternate time base. In it, the new trajectory starts at time 13.418
ros.joint_trajectory_controller: Trajectory has 3 segments:
- 1 segment(s) will still be executed from previous trajectory.
- 1 segment added for transitioning between the current trajectory and first point of the input message.
- 1 new segments (2 points) taken from the input trajectory.
ros.joint_trajectory_controller.actionlib: Accepting goal, id: /basketball_robot-3-1462838727.239, stamp: 1462838727.24
```

andreaskoepf, 2016-05-10:
While fixing our MoveIt! problem I noticed that the messages published on `/joint_states` when runnning the ur_modern_driver with enabled ros_control had 'wrong' time-stamps - e.g. uptime of the machine instead of 'official' ROS-time. 

I think passing a ros-time time constructed from `clock_gettime(CLOCK_MONOTONIC, &current_time);` is not OK - at least until ros_control uses this time as the stamp for the generated joint_state messages. I think this should be changed to ros::Time::now() at [ur_ros_wrapper.cpp#L574](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L574).

Publishing ros-messages with 'uptime of the machine' time-stamps instead of the official ros time makes it really hard (actually impossible) to interpret those messages in other procsses - e.g. I tried to ensure that the received joint-states were newer than the planning request in MoveIt! but this did not work since the ur_modern_driver publishes (indirectly via ros_control) time-stamps with a different base (e.g. machine uptime).

In [joint_state_controller.cpp#L91](https://github.com/ros-controls/ros_controllers/blob/5d3de7f2b612d834c81a8f84ea0a210dbaba17fc/joint_state_controller/src/joint_state_controller.cpp#L91) seems to be the place where the ros-time passed form outside is set to the `sensor_msgs::JointState`.

andreaskoepf, 2016-05-10:
@ThomasTimm I just read the discussen at https://github.com/davetcoleman/ros_control_boilerplate/issues/4 - could you change the impl to match https://github.com/davetcoleman/ros_control_boilerplate/pull/5/files ? I think that would fix the problem.

ThomasTimm, 2016-05-10:
Made the fix a week ago, just waited with the push to make sure Dave reached the same conclusion :)

commit 5b22e464 (2016-05-10): Changed time base for ros_control. Fixes #44
closed 2016-05-10 by ThomasTimm