ros-industrial-attic/ur_modern_driver #45: Life threatening moves

andreaskoepf, 2016-05-08:
I am currently analyzing why in about 1 of 100 Trajectories generated by MoveIt! our UR completely turns mad and moves with insane maximum speed to invalid positions: Most of the time the UR itself stops the action with a "Protective Stop: Position deviates from path." error. As you can imagine I had some VERY SHOCKING moments over the last days when without advance notice the robot turns to maximum speed (actually accelerations I have never observed under normal conditions)...

I am analyzing three parts now which are involved: MoveIt!, the [ros_controllers](https://github.com/ros-controls/ros_controllers) position-controller and the UR-Modern-Driver. I wrote a debugging-tool that analyzes the trajectory goal message as they are generated by MoveIt! and currently believe that at least part of the problem is MoveIt! setting invalid (e.g. old) states as start-state for a plan (e.g. observed by others in https://github.com/ros-planning/moveit_ros/issues/442).

My question/suggestion regarding the UR-Modern-Driver is now: Would it be possible to install a line of defense against invalid movements by adding a sanity check before sending servoj-commands to  the UR? E.g. if the desired joint position are absolutely our of reach within the next 2-3 time-steps I think it would be reasonable to refuse the execution. This would be a great step towards a safe ROS-controlled UR5 (currently it really is life threatening if errors occur that I regularly saw over the last days...).

I could prepare a PR which adds the sanity check...

My testing environment: Ubuntu 14.04, ROS Indigo, UR5 CB3

andreaskoepf, 2016-05-09:
I noticed when not using ros-control there is already a check for an invalid start position in [ur_ros_wrapper.cpp#L330-L336](https://github.com/ThomasTimm/ur_modern_driver/blob/f104b500262cf7ffc272ec0c237e45e47826d4f0/src/ur_ros_wrapper.cpp#L330-L336). This is great! I wish something similar had been added to the ros_controllers Trajectory controller.

ThomasTimm, 2016-05-09:
The ros_controller is purely velocity based, and thus none of the desired poses (start, end , intermittent) is known to the driver. Therefore it is not possible to implement this check in the driver.

The driver must follow the commanded velocities as closely as possible, and thus such a check should be performed before the command even arrives at the driver. It is impossible for the driver to know when high velocities are life threatening and when they are desired.

We have a parameter for max_velocity, but that is currently not used with the velocity interface, only the action interface. I guess it would make sense to change that so it is also evaluated for incoming velocities. This would prevent high velocities both at the beginning of a trajectory as well as later in the execution though.

If the issue is with MoveIt!, then you should get the people behind that package to look into it.
An alternate solution could be to have a filter between MoveIt! and ros_control that discards trajectories that you see as life threatening.

andreaskoepf, 2016-05-09:
@ThomasTimm I think a max_velocity (e.g. per joint would be already perfect) constraint somewhere at a low level would be great - e.g. ignoring a servoj operation if the difference to the target joint-position exceeds a certain threshold.
The core of the problem is MoveIt! which seems to have a race-condition that results in using the start point of the previous trajectory .. this leads to the situation that a joint_trajectory-action message is generated with a first way-point which is possibly far away from the current robot state. The ros_controller could detect this (e.g. in the same way as the invalid start-position protection which is part of your driver when not using ros-control). But instead of detecting the invalid start-point it tries to blend from the current robot state to the 2nd point of the invalid trajectory. And this blending is done with nice quintic-spline interpolation which makes the first few steps of the trajectory acceptable for the UR and does not lead immediately to a "path deviation" error.

For safety reasons (e.g. safe co-bot scenarios) I think it would be desirable to let every part of the stack, e.g. path-planner, controller, robot-driver and hardware run some form of action-validation/sanity check (e.g. an option to limit the maximum velocity/acceleration).

andreaskoepf, 2016-05-09:
Just for reference: I commented about the ros_controller problem [here](https://github.com/ros-controls/ros_controllers/issues/68#issuecomment-217766962) in an old issue regarding the "Dropping all 1 trajectory points, as they occur before the specified time.." warning which ros_controller emits basically for each plan generated by MoveIt!

miguelprada, 2016-05-09:
> The ros_controller is purely velocity based, and thus none of the desired poses (start, end , intermittent) is known to the driver. Therefore it is not possible to implement this check in the driver.

This is not quite true. First, there's several versions of the joint trajectory controller in ros_control, which use different hardware interfaces under the hood: `PositionJointInterface`, `VelocityJointInterface` and `EffortJointInterface`. In addition, whichever the hardware interface being used, the controller has read access to the joint's position, velocity and effort, through the joint handle.

That being said, my (limited) understanding of ros_control architecture says that the joint limit enforcement is not performed by the controllers or the controller manager, and that it has to be implemented at the `RobotHW` level, by using the utilities in the `joint_limits_interface` package. As an example, see how this is done at the beginning of the `write()` funciton in both [`SimHwInterface`](https://github.com/davetcoleman/ros_control_boilerplate/blob/jade-devel/src/sim_hw_interface.cpp#L66) in `ros_control_boilerplate` and in [`LWRHWFRI`](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_hw/include/lwr_hw/lwr_hw_fri.hpp#L81) in `kuka-lwr`.

I currently don't have the time to work on a sensible PR, but the addition of such check (implementing limit enforcement at the RobotHW layer) would at least make sure that crazy commands do not reach the robot.

ThomasTimm, 2016-05-09:
> >The ros_controller is purely velocity based, and thus none of the desired poses (start, end , intermittent) is known to the driver. Therefore it is not possible to implement this check in the driver.

> This is not quite true. First, there's several versions of the joint trajectory controller in ros_control, which use different hardware interfaces under the hood:  PositionJointInterface , [...]

What I meant to say is that the ros_control interface on **this** driver is purely velocity-based. The ros_control config files I've supplied configures a velocity-based position controller, but is it also possible to make a velocity-based velocity controller, if it is desirable to have the robot move with desired velocities, i.e. for plasma cutting, welding or the like.
I have also made a position-based interface and supplied a position controller configuration, but not documented it anywhere as it at one point was very unstable.
Very fast acceleration have a tendency to trigger a protective stop on the robot when using the position-based controller, so I think @andreaskoepf is using the velocity-based position controller, and thus the servoj command is not even used.
It is currently not possible to use an effort-based interface on the UR, as it is not possible to command a force or acceleration directly to the robot joints.

It is correct that you can enforce joint limits, and that is also what I suggested above. But these are fixed limits ("don't move faster than 2 rad/sec" or "the max joint angle is 2 rad"). This doesn't change the fact that the driver doesn't know anything about the trajectory when controlled using the ros_control interface.
In the ros_control architecture, the desired trajectory is planned by MoveIt! (or another node), which publishes the trajectory to the ros_control-based controller. This controller calculates the error between the trajectory and the current position of the arm and then commands a target velocity of the arm to minimize this error. How fast the target velocity changes, and thus how fast the arm accelerates, is dictated by the P, I, and D parameters of the controller. Then all that is sent to the ur_modern_driver is this target velocity.

andreaskoepf, 2016-05-09:
@ThomasTimm I am pretty sure I am using the position trajetctory controller, e.g. my debug trajectory tool subscribes to the topic `/pos_based_pos_traj_controller/follow_joint_trajectory/goal` and the launch file contains:

``` <node name="ros_control_controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="joint_state_controller pos_based_pos_traj_controller" />````

 We also tried the velocity interface before but with a P value of 2 we got a unstable controller behaviour. I recently read in https://github.com/ThomasTimm/ur_modern_driver/issues/27 that a value of 1.2 might be stable... we will try this soon.

ThomasTimm, 2016-05-09:
okay, then you are probably using the position-based. Then I would recommend that you change to the velocity-based. But that is not really related to this issue.
As I wrote above, I had some problems with the position-based and thus thought I left out any mentions of it, but apparently it was only in my report I left it out. I can see it is described on the main page here on github :P
When time permits I will include a velocity check on both the position-based and the velocity based interface, but until then (and maybe also afterwards)  I would suggest you set up a filter between MoveIt! and ros_control to stop any trajectories that you find dangerous. For some people a high velocity is desirable, while you find it life threatening. It is impossible for me to write software that knows when the user want high speed and when they don't ;)

closed 2019-11-23 by gavanderhoorn