ros-industrial-attic/ur_modern_driver #47: Overshoot with position-based ros_control interface (servoj)

miguelprada, 2016-05-09:
I've had similar experience to what @andreaskoepf describes in #46. Working with the `ros_control` version of the driver, I switched to using the `PositionJointInterface`-based joint trajectory controller because I was suffering from pretty big stability issues when using the `VelocityJointInteface`-based joint trajectory controller (I reckon that this can be alleviated by tuning the PID parameters in the latter, but I figured that using the former was a quicker way to avoid the issue altogether).

When using the position interface in the driver, I see pretty noticeable overshoot, whether I use the `joint_trajectory_controller` in `ros_controllers` or other custom position controllers I've implemented myself. I don't think this comes from the controllers themselves, since the same controllers don't show this behavior when used to control a LWR. Since the `ur_modern_driver` does nothing more than forwarding the command from the controller to the servoing thread in the UR, I'm guessing this is some artifact of using `servoj`.

Has anyone else seen this effect and played a bit with the parameters to `servoj` to try to improve its performance?

ThomasTimm, 2016-05-09:
I haven't noticed it, but at some point I will try and look into it.
This probably won't be the highest prioritized task on my to-do list though, as the velocity-based interface works better on all metrics (i.e. reacts faster and doesn't have this issue) if one bothers to tune the controller.
Alternatively I would appreciate if anybody else would have a look at this.

alemme, 2016-05-11:
Hello everyone,
I recorded the 6DOF joint values in the `void UrDriver::servoj(std::vector<double> positions, int keepalive)` during one planed trajectory execution with ros position controllers.
I plotted the norm of the joint velocity vector.
As one can see in the figures below, there is no overshoot. However, the robot is not behaving accordingly to these commands as we still observe a slight overshoot at the endeffector of the real ur5 robot.
[image]
[image]

miguelprada, 2016-05-12:
@alemme I reckon that what you're recording are the _desired_ joint positions, which are sent through `UrDriver::servoj`. If there's no overshoot there, that reinforces my belief that the overshoot is caused by the servo control loop in the UR itself.

I think it would be good to replicate your experiment while also recording the measured joint states, and try to see what's the difference between them. Our robots are away right now, but I'll try to clear some time to do this test when they're back.

Anyways, if my hypothesis is true, there's probably not much we can do about it. I'm not sure what the lookahead_time` parameter does, but my intuition tells me that increasing its value will probably reduce the overshoot at the cost of increasing the reaction time to the commands.

I've skimmed through your report, @ThomasTimm, and I see that the performance is at its best when using `speedj` via `ros_control`, so I'll consider switching to that and see if it improves my issues.

ThomasTimm, 2016-05-12:
Well, if it depends on the lookahead time, that would explain why I haven't noticed it. I only have very limited access to a real robot with firmware 3.x, so I've mostly been experimenting with the simulator or a real robot with firmware 1.8, where the lookahead time is fixed to 0.1 sec.

@andreaskoepf / @alemme Try to change the lookahead value in https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp#L185 from 0.03 to 0.1 and see if that solves the problem. 
If it does, I can make it possible to set the value with a ros_param.

alemme, 2016-05-13:
I have tried 0.1 as suggested, which was actually worse than 0.03. However, adjusting the gain parameter did the job. `sprintf(buf, "\t\t\t\tservoj(q, t=%.4f, lookahead_time=0.03, gain=1000)\n", servoj_time_);` 1000 is probably a bit much in the long run, but for now it stopped the overshoot.

andreaskoepf, 2016-05-13:
@alemme The UR now stops at a trajectory end-point 'as if it was nailed into the air' - the robot fells much more stiff and under control, great job.

Chunting, 2016-05-31:
Hi, @ Alemme,
I am a beginner for UR ROS, and I am interested in your work on recording the UR values, such as positions, velocities ....
Would you please give me some tips on how you did it?
I am not very familiar on the real time communication with UR5, but I need recode the robot states at rate of 125Hz. If you can give me some examples, I will appreciate it.

Thanks,
Chunting

alemme, 2016-05-31:
@Chunting In this issue I just dumped the desired joint values in a file (directly in ur_driver.cpp). In general I would recommend to use  [rosbag](http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data) to record data.

srsidd, 2016-08-26:
Hi @alemme, I am using ros control with the position based controller on a UR5 and am running into the same problem - the robot overshoots and then seems to stabilize. I did however adjust the gain parameter as you recommended. What I did notice however was that changing the values of the gains in the ur5_controllers.config file don't have any effect on the robot's motion. Did you try playing around with those gain values?

miguelprada, 2016-08-26:
@srsidd the gains for the `pos_based_pos_traj_controller` are not used at all AFAIK, so what your say makes sense. The gain in the `servoj` command should make a difference, though, as @alemme suggests.

srsidd, 2016-08-26:
Thanks for your reply @miguelprada, I was trying to understand how to tune the parameters and came across this. Also @alemme  mentions "1000 is probably a bit much in the long run" I want to understand what exactly he means.

alemme, 2016-08-28:
@srsidd I did not try to find a better parameter yet since 1000 did the job and no further adjustments for our task was needed. However, if you want to have e.g. a smoother stopping and starting motion then maybe a smaller gain is better suited.

miguelprada, 2016-08-29:
Given this seems to boil down to the internals of `servoj` and not the ROS driver, I think it should be safe to close this issue.

closed 2016-08-30 by ThomasTimm

miguelprada, 2016-08-30:
Nice! Thanks, @ThomasTimm!

eirik-strand, 2017-11-13:
Hi, I think Im having the same issue only with velocity based "out of the box" planning with moveit.  As In the gif below you can see the arm trying to reach a goal(orange) for then to overshoot the position. Im seeing this problem all the time. The planning goes well, so I suspect that the problem is with the driver.  

[image]

I have tried to adjust the /ur_driver/servoj_time and /ur_hardware_interface/servoj_gain parameters, but it does not make any difference. 
Im using the simulator from ur, but the real robot gives me the same error. 
How to reproduce: 

```
roslaunch ur10_moveit_config moveit_rviz.launch config:=true

roslaunch ur_modern_driver ur10_bringup.launch limited:=true robot_ip:=ur_ip [reverse_port:=REVERSE_PORT]

roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch limited:=true

```

commit e785ce08 (2016-08-30): Added the servoj gain and servoj lookahead time as a parameter at launch time. Resolves #47