ros-industrial-attic/ur_modern_driver #48: Wrong length of message on RT interface

andreaskoepf, 2016-05-09:
From time to time (if the robot is running multiple hougs) I see log messages like:

```
Wrong length of message on RT interface: 1072623616
Wrong length of message on RT interface: 1072473427
Wrong length of message on RT interface: 1072398338
Wrong length of message on RT interface: 1072323238
Wrong length of message on RT interface: 1072473427
Wrong length of message on RT interface: 1072097960
Wrong length of message on RT interface: 0
```

Is this something to worry about?

sepjansen, 2016-05-09:
I also see these messages sometimes.

It happens especially when I'm running the UR simulator at higher speeds, e.g. at frequency of 250 Hz rather than the standard 125 Hz (You can do this only in simulation of course) . That would confirm your suspicions.

I never had any issues with it though.

ThomasTimm, 2016-05-10:
As noted in #42 this can safely be ignored. Apparently UR broadcast something undocumented on the realtime port from time to time. When I implement official support for firmware 3.2 ( See #17 ) this message will disappear.

closed 2016-05-10 by ThomasTimm