ros-industrial-attic/ur_modern_driver #49: Can this be bloomed, or can I bloom it?

TheDash, 2016-05-10:


ThomasTimm, 2016-05-11:
This package will be moved to ros-industrial in the near future ( #18 ) to replace the old ur_driver.
I fear creating an independent bloom release now will only create confusion.

TheDash, 2016-05-11:
It seems like that issue has been open for 7 months now. I can change the bloom stats later and transfer ownership, but I'd really prefer the *.deb available to our platforms and all via a public distro rather than by source.

ThomasTimm, 2016-05-11:
Yea, it's only three weeks since I was given write access to the Universal Robot repo here on github, and the same day I was informed that my PhD thesis was accepted for public defense, so time to fix the last issues before transferring has been limited.
But hopefully the transfer will be completed in the next month time or so.

closed 2016-08-26 by ThomasTimm

gavanderhoorn, 2016-12-30:
Not sure this was clear, but apparently there is a `0.0.3` version released into Indigo of `ur_modern_driver`: [status_page/ros_indigo_default.html?q=modern](http://repositories.ros.org/status_page/ros_indigo_default.html?q=modern).