ros-industrial-attic/ur_modern_driver #50: Expose digital IO from realtime interface

sepjansen, 2016-05-12:
At the moment, the IO states of the robot are communicated over a 'slow' interface of 10 Hz.
The real-time interface of UR (port 30003) communicates at least the 8 digital input pins at 125 Hz as well. From firmware 3.2 onward, the 8 digital output pins are also available on this interface.

I'd like to expose this data as a separate ROS message (e.g. /io_states_rt) with the same message type as /io_states (i.e. ur_msgs/IOStates.msg). This seems neater than increasing the publishing frequency of /io_states, since not all IO states are available through the real-time interface, like the analog and tool pins.

Btw: this would involve extending the real time interface data stream for firmware 3.2, as mentioned in #17, so I'd like to do that at the same time.

Suggestions welcome.

NikolasE, 2016-12-16:
I implemented an rtde interface: https://github.com/NikolasE/ur_modern_driver/blob/master/scripts/rtde_example.py 

It runs at 125Hz

sepjansen, 2016-12-23:
Great to have a working RTDE example! I think in future it would be great if the ur_modern_driver supports this interface.

For now, I've expanded the existing real-time interface on port 30003 to include at least the digital IO's. See #91.

gavanderhoorn, 2019-06-29:
Closing as the `kinetic-devel` refactor does exactly what the OP requested.

RTDE is a different topic and will be supported by the new driver.

closed 2019-06-29 by gavanderhoorn