ros-industrial-attic/ur_modern_driver #51: problem with test_move.py and ur5_bringup.launch on CB3.2 Ur5

Chunting, 2016-05-23:
Hi, 
When I run the rosrun ur_modern_driver test_move.py , it always stops at the "waiting for the server" and never continue. I program on UR5, CB 3.2.18744. UR5 IP is 192.168.1.102, and my PC is 192.168.1.100.
`2016-05-23 17:08:17 ☆  Chunting in ~/catkin_ws
○ → source devel/setup.bash 

 2016-05-23 17:08:29 ☆  Chunting in ~/catkin_ws
○ → rosrun ur_modern_driver test_move.py
Waiting for server...
^Z
[1]+  Stopped                 rosrun ur_modern_driver test_move.py`
 

What's more , when I run the command roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102 [reverse_port:=REVERSE_PORT]

` 2016-05-23 17:15:59 ☆  Chunting in ~/catkin_ws
○ → roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102 [reverse_port:=REVERSE_PORT]
... logging to /home/chunting/.ros/log/bc99c3ea-20c5-11e6-bf75-44850011896a/roslaunch-Chunting-13638.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Chunting:44301/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.19
 * /ur_driver/base_frame: base_link
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 192.168.1.102
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [13659]
process[ur_driver-2]: started with pid [13660]
[FATAL] [1463995069.901632143]: Error connecting to get firmware version
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[ur_driver-2] process has died [pid 13660, exit code -6, cmd /home/chunting/catkin_ws/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/chunting/.ros/log/bc99c3ea-20c5-11e6-bf75-44850011896a/ur_driver-2.log].
log file: /home/chunting/.ros/log/bc99c3ea-20c5-11e6-bf75-44850011896a/ur_driver-2*.log

It does not succeed.

I can ping the robot.
Would you please tell me how correct it?

Thanks

Chunting 
jiaochunting@gmail.com

Chunting, 2016-05-23:
I have tried for several times but it does not work. Thanks for your help.

ThomasTimm, 2016-05-24:
The test_move script just tries to send some trajectories to the driver. So when it outputs "waiting for server", it is referring to the action server in the driver, which it is waiting for.

So, before doing `rosrun ur_modern_driver test_move.py`, you need to start the driver with `roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102`. Note that you don't need to supply a reverse port unless you have some special network setup, and even if you do, you obviously need to specify a port number, not the string "REVERSE_PORT".

Also, trying the same thing several times usually yields the same result each time...

Chunting, 2016-05-24:
Thanks for your anwser, Thomas.
I've tried the port 30001as below
`roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102  reverse_port:=30001`

It does not work either.
The information is 

```
... logging to /home/chunting/.ros/log/b0494ca8-21a8-11e6-bfea-44850011896a/roslaunch-Chunting-4023.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Chunting:34763/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.19
 * /ur_driver/base_frame: base_link
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 192.168.1.102
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[robot_state_publisher-1]: started with pid [4044]
process[ur_driver-2]: started with pid [4045]
[FATAL] [1464092328.601440595]: Error connecting to get firmware version
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
[ur_driver-2] process has died [pid 4045, exit code -6, cmd /home/chunting/catkin_ws/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/chunting/.ros/log/b0494ca8-21a8-11e6-bfea-44850011896a/ur_driver-2.log].
log file: /home/chunting/.ros/log/b0494ca8-21a8-11e6-bfea-44850011896a/ur_driver-2*.log
```
It seems that the main problem is 
```
[FATAL] [1464092328.601440595]: Error connecting to get firmware version
terminate called after throwing an instance of 'std::system_error'
```

Thanks for your help.

Sincerely,
Chunting

Chunting, 2016-05-24:
Hi, Thomas,
As is shown above, but I did not find the log **ur_driver-2*.log**, which does not exit.
In the master.log, the information is as below:

```
[rosmaster.main][INFO] 2016-05-24 20:47:11,652: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2016-05-24 20:47:11,652: Starting ROS Master Node
[xmlrpc][INFO] 2016-05-24 20:47:11,653: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2016-05-24 20:47:11,653: Started XML-RPC server [http://Chunting:11311/]
[rosmaster.master][INFO] 2016-05-24 20:47:11,653: Master initialized: port[11311], uri[http://Chunting:11311/]
[xmlrpc][INFO] 2016-05-24 20:47:11,653: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2016-05-24 20:47:11,671: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:47:11,672: +PARAM [/roslaunch/uris/host_chunting__41979] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:47:11,715: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:47:11,715: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:47:11,719: +SERVICE [/rosout/get_loggers] /rosout http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:47:11,720: +SERVICE [/rosout/set_logger_level] /rosout http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:47:11,721: +PUB [/rosout_agg] /rosout http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:47:11,724: +SUB [/rosout] /rosout http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:48:06,745: +PARAM [/roslaunch/uris/host_chunting__42811] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/rosdistro] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/tool_frame] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/robot_ip_address] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/max_payload] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/min_payload] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/prefix] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/max_velocity] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/rosversion] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/base_frame] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/robot_description] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,749: +PARAM [/ur_driver/servoj_time] by /roslaunch
[rosmaster.master][INFO] 2016-05-24 20:48:06,805: +PUB [/rosout] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,805: +SERVICE [/robot_state_publisher/get_loggers] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,806: +SERVICE [/robot_state_publisher/set_logger_level] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,809: +PUB [/rosout] /ur_driver http://Chunting:44585/
[rosmaster.master][INFO] 2016-05-24 20:48:06,810: +PUB [/tf] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,810: +SERVICE [/ur_driver/get_loggers] /ur_driver http://Chunting:44585/
[rosmaster.master][INFO] 2016-05-24 20:48:06,811: +PUB [/tf_static] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,811: +SERVICE [/ur_driver/set_logger_level] /ur_driver http://Chunting:44585/
[rosmaster.master][INFO] 2016-05-24 20:48:06,817: +SUB [/joint_states] /robot_state_publisher http://Chunting:35307/
[rosmaster.master][INFO] 2016-05-24 20:48:06,856: publisherUpdate[/rosout] -> http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:48:06,857: publisherUpdate[/rosout] -> http://Chunting:44360/
[rosmaster.master][INFO] 2016-05-24 20:48:06,958: -PUB [/rosout] /ur_driver http://Chunting:44585/
[rosmaster.master][INFO] 2016-05-24 20:48:06,959: -SERVICE [/ur_driver/get_loggers] /ur_driver rosrpc://Chunting:37920
[rosmaster.master][INFO] 2016-05-24 20:48:06,960: -SERVICE [/ur_driver/set_logger_level] /ur_driver rosrpc://Chunting:37920
[rosmaster.master][INFO] 2016-05-24 20:48:07,057: publisherUpdate[/rosout] -> http://Chunting:44360/
```

Thanks,
Chunting

Chunting, 2016-05-25:
I use my laptop to run the command, just after I disconnected wifi, everything goes well...

ThomasTimm, 2016-05-30:
Good to hear. You should probably look into defining your default network interface...

When using a Universal Robot it causes unnecessary confusion using port 30001 - 30004 as the reverse port. 
Just launch the driver with `roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102`

closed 2016-05-30 by ThomasTimm

haniehes, 2017-09-22:
Hi, I had the similar problem as **chunting**. 
 I have done similar procedures to communicate with UR5 through ROS Kinetic, but I could not solve it. After `roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=192.168.1.102` 
I encounter: `[FATAL] [1464092328.601440595]: Error connecting to get firmware version
terminate called after throwing an instance of 'std::system_error'`. 
I have to mention that I use my laptop to run the command, but after I disconnected wifi and connected to wired connection the error still and exists!
I appreciate if you can you guide me in this regard?

Chunting, 2017-09-28:
Hi,
I also used my laptop to do my work, and after I physically disabled wifi
connection, everything  returned to normal.
Hope this can help you.

Thanks,
Chunting