ros-industrial-attic/ur_modern_driver #53: tool0_controller/tool0 frame offset

achim-k, 2016-05-31:
When I run `roslaunch ur_modern_driver ur10_bringup.launch robot_ip:=...`  I get a weird position of the `tool0_controller` frame. It seems like that the `tool0_controller` frame is the mirrored version of `tool0` (symmetrical around `base` 

[image]
[image]

ThomasTimm, 2016-06-02:
I haven't seen it before, no.
Did you follow what is written in the readme, including making sure you have a version of robot_state_publisher that is able to handle floating joint?
Is it possible for you to say which one is correct if you compare it to the info on the teach pendant?
Have you changed anything in the URDF or using any joint offsets?

gavanderhoorn, 2016-06-02:
The orientation of the `base_link` in the UR5 and UR10 models is the mirror (rotated `pi` over Z) of the orientation of the real robot's *Base* feature. See also ros-industrial/universal_robot#200. Afaik, this has been the case since the origin WG developers created the package.

The `base` and `tool0` frames added in ros-industrial/universal_robot#200 have the correct orientation, and correspond directly to the *Base* feature and an all-zeros TCP.

A real fix would be to fixup the urdfs in `ur_description`, but I've seen people just specify the `base` frame for the `base_frame` parameter in `launch/ur_common.launch`.

achim-k, 2016-06-06:
Thanks @gavanderhoorn, manually setting the 'base_frame' param to 'base' works for me. Shouldn't this then become the default?

ThomasTimm, 2016-06-13:
I agree. Anybody has a good reason why this shouldn't be the default value?

gavanderhoorn, 2016-06-13:
I've not been following the development of `ur_modern_driver` close enough to know why it was set t `base_link` in the first place. It seems like an odd choice, as that frame has always had that problem. Perhaps knowing that we (or you, @ThomasTimm) can derive / tell us why it was set to `base_link`, and then why it should, or shouldn't be changed?

ThomasTimm, 2016-06-13:
When @sepjansen added the tool0 frame, the parent frame was correctly set to `base`. Apparently, I accidentally changed that in commit adc223
So it should obviously be changed back, thanks for noticing this @achim-k

commit 091f7c3b (2016-06-14): Chnaged tool0_controller base frame. fixes #53
closed 2016-06-14 by ThomasTimm