ros-industrial-attic/ur_modern_driver #54: Failure due to protective stop on the UR

VictorLamoine, 2016-06-02:
The robot stops during the execution of motions programmed via MoveIt;
Example package:
https://github.com/InstitutMaupertuis/universal_robot_motion

See:
https://github.com/ros-industrial/universal_robot/issues/241#issuecomment-223309926

VictorLamoine, 2016-06-09:
General information :
- ROS Indigo, Ubuntu 14.04 64 bits.
- The end effector mass is 0.5 kg and it is declared in the UR10 controller.
- I'm using `universal_robot` `indigo-branch`.

These are my test pacakges:
https://github.com/InstitutMaupertuis/institut_maupertuis_robots_descriptions/tree/ur10_tests
https://github.com/InstitutMaupertuis/universal_robot_motion
https://github.com/InstitutMaupertuis/ensenso_extrinsic_calibration/tree/ur10_test

[kinematics.yaml file](https://github.com/InstitutMaupertuis/institut_maupertuis_robots_descriptions/blob/ur10_tests/ur10_ensenso/ur10_ensenso_moveit_config/config/kinematics.yaml)
[joint_limits.yaml file](https://github.com/InstitutMaupertuis/institut_maupertuis_robots_descriptions/blob/ur10_tests/ur10_ensenso/ur10_ensenso_moveit_config/config/joint_limits.yaml)

`universal_robot_motion`
---
This program moves the robot using the `tool0` TCP (`manipulator` move group).
The robot does not go into protective stop with this trajectory/setup however the robot motion is very jerky at high speeds on the teach pendant.

- [Video 1, speed = 24%](https://mega.nz/#!1BUgUYDZ!-KKiZ95A7is_akwBfdUWyBWrUNOt-HTAef0DLkHNQLI)
- [Video 1, speed = 100%](https://mega.nz/#!YRlg0ZoT!uj2BSeI8OFckhJ1IfJesUXz2YkagipYXrs6Z09wWFaQ)

```
$ rostopic echo -n1 /joint_states
header: 
  seq: 30115
  stamp: 
    secs: 1465464907
    nsecs: 757682323
  frame_id: ''
name: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
position: [0.10749213397502899, -1.6914003531085413, -2.255641285573141, -0.7656462828265589, 1.570691466331482, -1.4633196035968226]
velocity: [0.003847735235467553, 0.0005752427969127893, -0.0010865697404369712, -0.017736652866005898, 0.0, 0.00046019424917176366]
effort: [0.3799438774585724, 3.437805414199829, 2.176798105239868, 1.029556393623352, -0.1547699123620987, 0.1682281643152237]
---
```

```
$ rostopic echo /joint_trajectory_action/goal
header: 
  seq: 27
  stamp: 
    secs: 1465464915
    nsecs: 547241536
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1465464915
    nsecs: 547242262
  id: /move_group-28-1465464915.547242262
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: /base_link
    joint_names: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
    points: 
      - 
        positions: [-2.2549708525287073, -1.6918318907367151, 0.10376204550266266, -0.7655866781817835, 1.5708353519439697, -1.467086140309469]
        velocities: [0.01, 0.01, 0.01, 0.01, 0.01, 0.01]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 200000000
      - 
        positions: [-2.2551142205139545, -1.6917144345912636, 0.104558172139388, -0.7655606542365021, 1.5708275466449029, -1.4662796425854197]
        velocities: [0.01, 0.01, 0.01, 0.01, 0.01, 0.01]
[... 588 lines cut ...]
      - 
        positions: [-1.8480427900897425, -1.9862878958331507, 0.0754837691783905, -0.9033015410052698, 1.5345759391784668, -1.7015331427203577]
        velocities: [0.05, 0.05, 0.05, 0.05, 0.05, 0.05]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 200000000
      - 
        positions: [-1.8507369204866926, -1.9846223054378545, 0.0760957076899158, -0.9010131990373642, 1.5349473955385382, -1.6905444304981287]
        velocities: [0.05, 0.05, 0.05, 0.05, 0.05, 0.05]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 400000000
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0
---
```

[Here](https://mega.nz/#!UVUT3YrQ!fpYUKsTOw1gaM19AHNUDa9ivqOMB5ADEvTy6Yg_PdZM) is a video of the robot going in protective stop.

ThomasTimm, 2016-06-09:
From the looks of it, my initial guess would be that your problem is due to the velocity values in your goal.
In the first example, each goal pose has a velocity of 0.01, spaced 0.2 sec apart. Obviously, this will result in jerky movement, as the robot first accelerates to be able to move sufficiently fast, and then slows down to get the velocity down to 0.01 rad/sec. The result is the jerky motion you see.
The same goes with the other example; here the velocity is a bit higher, but the travel distance is also much higher.
To visualize the problem, try to plot the velocity and acceleration curve required to execute your trajectory, interpolating with a timestep of 8 ms. You can find the code used by the driver for interpolation at https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp#L61

gavanderhoorn, 2016-06-09:
@VictorLamoine: what did you use to calculate velocities and accelerations?

VictorLamoine, 2016-06-13:
Argh, I didn't notice that the trajectory already had timing information!
I never played with timed/velocity constrained trajectory before.

Some things are not clear to me:
- How does it work when we specify both the goal time and velocity, these information are contradictory?!
- What should I put in velocities?
 - Per-axis maximum/goal velocities?
 - TCP maximum/goal velocity?
- If I want a specific TCP speed, what should I do?
- Can I put all `time_from_start` to zero If I want to ignore this information?

Where is that documented? I can't find the right wiki link.

ThomasTimm, 2016-06-13:
The velocity is related to the point at the trajectory, so you should use the desired joint velocity at the point (i.e. all zeros at the first and last point).
The driver then interpolates and calculate the required velocity to reach the trajectory points at the desired time.
So goal time and velocity is not contradictory; goal time is when the robot should be at that pose, velocity is how fast the robot should move at that point.
If you want a specific TCP speed in cartesian space, you should find and calculate the inverse Jacobian.
No, if you put all time_from_start to zero, you are essentially telling the driver at should be at all those poses at the same time (now).

VictorLamoine, 2016-06-13:
Thanks, I'll try that Thursday and give you feedback!

VictorLamoine, 2016-06-16:
I think understood how it works but this feels completely un-usable to me.

All I want is a smooth motion at a given constant speed. I don't care about accelerations as long as it's not too slow. And I don't want to calculate anything, ROS should do that for me.

There are many parameters interacting with each other:
- Compute Cartesian path `eef_step`
- `time_from_start` (can we change how much it increases between each point? I don't want to manually tweak those values!)
- Velocities

If we change one of those without adjusting the others, it breaks everything.

It's hard to believe people are doing it this way.. this is way over-complicated for what I'm trying to achieve, am I using the right tool? Is there a package or anything I'm missing?

I'm frustrated because I'm trying to achieve something that should be very simple to achieve (10 secs of programming with the UR teach pendant) and it feels like a pain to do it with ROS-I.

I'm getting a lot less preventing stops because I understood how to tweak the parameters but it still happens sometimes. The robot is finishing the trajectory very brutally despite the fact that I set all the velocities of the last pose to zero, it seems like it stops before reaching the goal.

ThomasTimm, 2016-07-19:
Seeing your issue is with ROS and calculating stuff, I'll close the topic. Let me know if you have any issues with the driver not working when using it as intended.

closed 2016-07-19 by ThomasTimm