ros-industrial-attic/ur_modern_driver #56: Occasional Jerky motion

homesick-nick, 2016-06-10:
UR5 running ployscope 3.2.18744. very minimal additional end effector mass, ubuntu 14.04 VM on windows host and ros-indigo. Both with test_move.py and when using the moveit_commander interface to plan and execute trajectories to ee goal positions, the arm occasionaly exhibits the odd jolt during execution, particularly for longer paths, when many joints are moving. Here is an example video:

[Video](https://www.youtube.com/watch?v=8kAmDNfSgL8  )

You see the jolt at roughly 3 seconds. This was performed with the moveit_commander interface.

Here is the output on follow_joint_trajectory/goal, that corresponds with the above video. 

```
header: 
  seq: 9
  stamp: 
    secs: 1465575469
    nsecs: 202095307
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1465575469
    nsecs: 202096304
  id: /move_group-10-1465575469.202096304
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: /world
    joint_names: ['elbow_joint', 'shoulder_lift_joint', 'shoulder_pan_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
    points: 
      - 
        positions: [-0.8290494124041956, -2.5253475348102015, -0.8362420240985315, 0.7118555307388306, -1.584247891102926, -1.302772347127096]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, -0.3327029212965289, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [-0.8957262914938168, -2.4424336433522873, -0.8870695181020145, 0.814757601131566, -1.5826221988115585, -1.3017115409826472]
        velocities: [-0.120625527129188, 0.15, -0.0919523130137044, 0.18616072997544747, 0.0029410493153477525, 0.0019191105238146062]
        accelerations: [0.0, 0.0, 3.7659579135098096e-16, 0.0, -7.27300622046582e-16, -7.269083347639247e-16]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 552759276
      - 
        positions: [-0.9624031705834379, -2.359519751894373, -0.9378970121054974, 0.9176596715243015, -1.5809965065201912, -1.3006507348381986]
        velocities: [-0.12062552712918757, 0.15, -0.09195231301370407, 0.18616072997544697, 0.0029410493153477443, 0.001919110523814601]
        accelerations: [1.5565959375840504e-15, 0.0, 8.787235131522865e-16, -1.8076597984847034e-15, 6.982713631299419e-16, 7.080785451963737e-16]
        effort: []
        time_from_start: 
          secs: 1
          nsecs: 105518552
      - 
        positions: [-1.029080049673059, -2.2766058604364585, -0.9887245061089802, 1.020561741917037, -1.5793708142288236, -1.2995899286937498]
        velocities: [-0.12062552712918767, 0.15, -0.09195231301370417, 0.18616072997544697, 0.0029410493153477443, 0.001919110523814802]
        accelerations: [-1.9080853428449648e-15, 0.0, -1.2553193045032663e-15, 1.8076597984847034e-15, -6.982713631299419e-16, 1.8829789567548997e-17]
        effort: []
        time_from_start: 
          secs: 1
          nsecs: 658277829
      - 
        positions: [-1.0957569287626803, -2.1936919689785443, -1.0395520001124632, 1.1234638123097724, -1.5777451219374563, -1.298529122549301]
        velocities: [-0.120625527129188, 0.15, -0.09195231301370452, 0.18616072997544747, 0.0029410493153477525, 0.0019191105238146062]
        accelerations: [7.280851966118966e-16, 0.0, 0.0, 0.0, 7.27300622046582e-16, -7.269083347639247e-16]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 211037105
      - 
        positions: [-1.1624338078523013, -2.11077807752063, -1.0903794941159461, 1.226365882702508, -1.5761194296460888, -1.2974683164048524]
        velocities: [-0.1206255271291878, 0.15, -0.09195231301370432, 0.18616072997544728, 0.0029410493153477525, 0.0019191105238146062]
        accelerations: [0.0, 0.0, 7.280851966118966e-16, -7.029788105218311e-16, -7.27300622046582e-16, 7.269083347639247e-16]
        effort: []
        time_from_start: 
          secs: 2
          nsecs: 763796381
      - 
        positions: [-1.2291106869419224, -2.027864186062716, -1.1412069881194289, 1.3292679530952431, -1.5744937373547214, -1.2964075102604036]
        velocities: [-0.12062552712918784, 0.14999999999999997, -0.09195231301370418, 0.18616072997544722, 0.002941049315347748, 0.0019191105238148044]
        accelerations: [-1.255319304503268e-16, -5.0212772180130727e-17, -2.510638609006536e-16, 5.021277218013072e-16, 7.123937053056047e-16, -9.807182066431782e-18]
        effort: []
        time_from_start: 
          secs: 3
          nsecs: 316555658
      - 
        positions: [-1.2957875660315437, -1.9449502946048016, -1.1920344821229119, 1.4321700234879788, -1.5728680450633539, -1.2953467041159548]
        velocities: [-0.12062552712918787, 0.14999999999999997, -0.09195231301370425, 0.18616072997544697, 0.002941049315347744, 0.001919110523814601]
        accelerations: [0.0, 0.0, 0.0, -1.456170393223789e-15, -7.2730062204658e-16, -7.269083347639226e-16]
        effort: []
        time_from_start: 
          secs: 3
          nsecs: 869314934
      - 
        positions: [-1.362464445121165, -1.8620364031468872, -1.2428619761263948, 1.535072093880714, -1.5712423527719865, -1.2942858979715062]
        velocities: [-0.12062552712918784, 0.14999999999999997, -0.09195231301370438, 0.18616072997544703, 0.002941049315347748, 0.0019191105238146036]
        accelerations: [1.255319304503268e-16, 5.0212772180130727e-17, -4.770213357112419e-16, 1.657021481944314e-15, 7.422075387875573e-16, 7.367155168303555e-16]
        effort: []
        time_from_start: 
          secs: 4
          nsecs: 422074211
      - 
        positions: [-1.429141324210786, -1.779122511688973, -1.2936894701298778, 1.6379741642734496, -1.569616660480619, -1.2932250918270574]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.4364486758786933, -0.542731732991224, 0.3327029212965289, -0.6735689039632393, -0.010641338611542983, -0.006943747869278039]
        effort: []
        time_from_start: 
          secs: 4
          nsecs: 974833487
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0
---

```

ThomasTimm, 2016-06-13:
Hmmm, it looks strange - Almost like a few network packages are dropped/buffered and transmitted at the same timr.
I don't have any experience using ROS with WM on a windows host, so I don't know if windows might use all of the NIC's resources for a few miliseconds.
Could you perhaps try it on a non-WM machine? or set up another machine with wireshark that monitors the network and then plot the timestamp when data is sent from the computer to the controller. (It should be 28 bytes transmitted at 125 Hz from the computer to the controller).

homesick-nick, 2016-06-13:
Thanks for the advice. I will have a look into it. No experience of wireshark, but immediately I'm inclined to think I'll run into problems due to working on a university network. 

Do you have any experience of running ROS with VM on any other host, - mac for instance?

Thanks,

Nick

ThomasTimm, 2016-06-14:
Working with a Motoman robot, I had severe problems with package losses. I solved this by inserting an extra network card in the lab PC and having a direct network link between the PC and the robot. Our URs are on the same internal network as the Motoman initially was on (A network dedicated to robots and ethernet-connected sensors and actuators), and haven't had any issues with that. But it obviously depends on the network load.
Try connecting directly to the robot with a cable.

I haven't used VM very much, so I don't know if lags occurs on the network. I have always just used dedicated Ubuntu machines, or dual boot on the machines that required windows as well.
If you can't dedicate a full PC to running ubuntu, you could run the driver on a raspberry pi or similar, and then have Rviz/moveit running on your laptop with the laptop configured as the ros-master.

homesick-nick, 2016-07-13:
Thanks Thomas. Sorry for the late reply. I've now fully tested on a dedicated Ubuntu machine and on a separate dedicated network. As far as I can tell you were correct; the robot seems to move smoothly when controlled from a physical machine. I haven't been able to analyse the connection with wireshark, but I can say that by pinging the robot from both physical and virtual machines the connection seems almost twice fast from the physical machine, so I wonder if this has anything to do with it. Anyway thanks again for your help!

homesick-nick, 2016-07-14:
Hi Thomas,

Thank you so much for all your help. You were correct that it was down to
using a VM! I now have to set up a dedicated Ubuntu machine or a dual boot
system. I would really appreciate some advice on the hardware requirements
for the PC.

I have a free PC available to me which is 64-bit, dual core (3.3GHz) with
4GB of ram, which is currently running windows 7. For various reasons I
ideally need to have a good idea as to whether this hardware is going to
cut it before I go ahead with the Ubuntu instillation. I'm going to be
running matlab programs along side all the ros nodes required to control
the arm with moveit.

What would you suggest? Is it likely that I'm going to need something with
more cores or more ram to avoid having to switch PCs again further down the
line?

Any help you can offer is greatly appreciated!!

Thanks,

Nick

ThomasTimm, 2016-07-18:
Well, making suggestions is rather difficult for me, as I don't know your setup or future plans. But unless you are planning on handling large data later on (like images or point clouds), your current setup should suffice.

As an alternative, the driver itself uses very few resources and as such could be running on a cheap raspberry pi or similar. Then you could connect the RPI and your VM machine, as the moveit-part is not so timing critical.

closed 2016-07-19 by ThomasTimm

homesick-nick, 2016-07-25:
Thanks for the advice!