ros-industrial-attic/ur_modern_driver #60: Communication problems UR3

mgvargas, 2016-07-14:
Hello,

I'm having some communication problems while trying to send commands to the robot, sometimes even with the test_move.py file. I still don't know why sometimes it works and some others it can't connect to the server.

`rosrun ur_modern_driver test_move.py`
`Waiting for server...`

The same happens when I try to generate an actionlib on my code, it keeps trying to connect but does nothing more...

I am running the ur_modern_driver ur3_bringup.launch and it seems to work fine:

`Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.`
`started roslaunch server http://leela:48102/`
`
SUMMARY
========`

`PARAMETERS`
` * /robot_description: <?xml version="1.... `
` * /rosdistro: indigo `
` * /rosversion: 1.11.19 `
` * /ur_driver/base_frame: neck_base `
` * /ur_driver/max_payload: 3.0 `
` * /ur_driver/max_velocity: 10.0 `
` * /ur_driver/min_payload: 0.0 `
` * /ur_driver/prefix: neck_ `
` * /ur_driver/robot_ip_address: 192.168.102.62 `
` * /ur_driver/servoj_time: 0.008 `
` * /ur_driver/tool_frame: neck_tool0_contro... `

`NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)`

`ROS_MASTER_URI=http://localhost:11311`

`core service [/rosout] found
process[robot_state_publisher-1]: started with pid [5204]
process[ur_driver-2]: started with pid [5205]
[ INFO] [1468499587.988261117]: Setting prefix to neck_
`

Any ideas of why is this happening? Am I missing something?

Thanks

ThomasTimm, 2016-07-18:
When the test_move script is outputting `waiting for server...`, it is referring to the driver, not the robot.

You should launch the ur3_bringup.launch file, and then in a new console run the test_move script.

mgvargas, 2016-07-18:
I am running the ur3_bringup.launch in another console. I forgot to mention, I'm remotely connecting to another computer using SSH, I run the ur3_bringup.launch in the remote computer (the one connected to the robot) and the test_move in mine.

To do this, i added some stuff to my BASHRC file:
   - export ROS_MASTER_URI=http://remote.computer
   - export ROS_IP=my_IP

The communication seems to be working.

miguelprada, 2016-07-18:
@mgvargas, it used to be the case that the ROS_MASTER_URI had to carry the port number where the ROS master is listening for connections. Try appending the default port number and see if this works.

`export ROS_MASTER_URI=http://remote.computer:11311`

mgvargas, 2016-07-18:
@miguelprada Hi, I'm already doing that, sorry for not mentioning it before

`export ROS_MASTER_URI=http://remote.computer:11311`

Other launch files seem to be working fine.

miguelprada, 2016-07-18:
Could you then try to run the test_move script in the remote computer as well? This way we could rule out  (or not) connectivity issues between the computers.

mgvargas, 2016-07-18:
When I run test_move in the remote computer before trying to run it in mine, it works without problems. But after running in it my laptop it stops working as well, it just keeps waiting for the server. In order to run it again I have to restart the robot and (sometimes) my laptop.

mgvargas, 2016-07-18:
I managed to solve the problem, apparently it was a firewall issue, it was blocking my messages.

closed 2016-07-19 by ThomasTimm