ros-industrial-attic/ur_modern_driver #61: Controlling IO via Python Scripts

atomoclast, 2016-08-09:
Hello all,

I know that the ur_driver as well as the ur_modern_driver both have similar test_move.py scripts, so I thought to try and run the test_io.py example in the ur_driver folder for the universal_robot package withe the ur_modern_driver package, and I'm having no luck...

Is there a way to control IO with the ur_ modern_driver and Python?

Is there an edit that needs to be made to test_io.py? 

Thanks in advance!

michael-hart, 2016-08-15:
Hi,

I'd also like to be able to at least set digital IO from Python. At the moment I'm having to blast strings across the URScript node to get my intended behaviour.

I understand that in the original driver, there was a service call available which understood ur_msgs::setDigitalOutput messages, but although I can find a handler in this repo, I cannot find the service.

Would you like me to implement this and submit a pull request?

ThomasTimm, 2016-08-26:
It is (probably, i haven't tried) true that the test_io.py from the old driver probably doesn't work with the new. Toggling IO should be straight-forward though, as the IO is controlled with a service call to`/ur_driver/set_io`, which is of the type `ur_msgs::SetIO`. 
Call that service as you would with any other services; the srv has 3 data fields: `fun`, `pin `and `state`. `fun` or function, should be 1 for digital out, `pin` should be the digital out pin number and `state` should be non-zero for high or 0.0 for low.

The service handler is defined on https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L198 and implemented on https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L382

closed 2016-08-26 by ThomasTimm