ros-industrial-attic/ur_modern_driver #62: Robot randomly moves upon launching ur10_ros_control.launch

srsidd, 2016-08-12:
I tried running the ur10 using ros control with the given launch file. I noticed that the robot keeps moving randomly though there is no moveIt running or any action client to give it a move command. However if I use only the position based controller the robot doesn't move. Can anyone help?

Thanks!

miguelprada, 2016-08-17:
Hi @srsidd. It might have something to do with #55. Could you check that out? It should've been fixed by #57, but I might've overlooked something.

srsidd, 2016-08-26:
Hi @miguelprada I went through what #55 and it looks like they're very similar but I haven't debugged the problem much. I switched to the position based controller instead.

srsidd, 2016-09-23:
HI @miguelprada, thank you for your reply. My problem seems very similar to #55, but I haven't really done a lot of debugging. It seemed like the robot was getting

miguelprada, 2016-09-26:
@srsidd Did you manage to figure out what your problem was? Or at least are you successfully using the position-based controller? Can we close this?

sidd1, 2016-10-12:
Hi @miguelprada , I was not able to find out what the issue was, I ran it only two or three times again but found the same behaviour repeating. For now I use the position based controller which seems to work fine.

srsidd, 2016-10-12:
@miguelprada, I was not able to diagnose the problem. I only ran the velocity based controller 2-3 times again. However I switched to the position based controller and that seems to be working fine for now.

miguelprada, 2016-10-13:
Would you consider closing this issue then, @srsidd?

srsidd, 2016-10-21:
Sure, if I do get back to this and do some debugging I'll post it here. Thanks for the help.

closed 2016-10-21 by srsidd