ros-industrial-attic/ur_modern_driver #63: Cannot find "hardware_interface" in canSwitch

qqwan, 2016-08-18:
Hi, 

I'm trying to compile it by 'catkin_make' and am running into the following error. Any hint on where to start? Thanks!

~~~~~~~~~~~~~~~~~~~~~~~~
/home/qwan/src/universal_robot/ur_modern_driver/src/ur_hardware_interface.cpp: In member function ‘bool ros_control_ur::UrHardwareInterface::canSwitch(const std::list<hardware_interface::ControllerInfo>&, const std::list<hardware_interface::ControllerInfo>&) const’:
/home/qwan/catkin_ws/src/universal_robot/ur_modern_driver/src/ur_hardware_interface.cpp:180:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface
                      ^
In file included from /home/qwan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /home/qwan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /home/qwan/catkin_ws/src/ros_control/hardware_interface/include/hardware_interface/joint_state_interface.h:33,
                 from /home/qwan/catkin_ws/src/universal_robot/ur_modern_driver/include/ur_modern_driver/ur_hardware_interface.h:61,
                 from /home/qwan/catkin_ws/src/universal_robot/ur_modern_driver/src/ur_hardware_interface.cpp:58:

carlosjoserg, 2016-08-18:
Hi @qqwan 

Given the error, I assume you are using ROS Kinetic, where the ros control API has changed w.r.t to Indigo and Hydro where this driver runs perfectly.

Check #58 for some temporary fixes I've tried successfully on Kinetic, and let know if they work for you as well.

Best

qqwan, 2016-08-18:
Actually, I am still on Indigo, and Ubuntu14.04. Although the ros_control package is something I just downloaded. Do you see any possibility of incompatibility?

carlosjoserg, 2016-08-18:
If you downloaded the ros control package, note that the default branch is
kinetic-devel, hence the problem you are having.

The full ros controls packages are available via sudo apt-get install
ros-indigo-*, which I'd suggest you to do instead of downloading.

qqwan, 2016-08-19:
I think I had all the right version of things, but there was some problem in the compiler, so I deleted the 'build' and 'devel' folder in my catkin folder altogether and rebuilt every single package. Now it works! Thanks for the help!

cheers.

closed 2016-08-19 by qqwan

nevermore0127, 2018-04-19:
Could not find a package configuration file provided by
  "hardware_interface" with any of the following names:

    hardware_interfaceConfig.cmake
    hardware_interface-config.cmake

  Add the installation prefix of "hardware_interface" to CMAKE_PREFIX_PATH or
  set "hardware_interface_DIR" to a directory containing one of the above
  files.  If "hardware_interface" provides a separate development package or
  SDK, be sure it has been installed.

gavanderhoorn, 2018-04-19:
@nevermore0127: I don't think this is the same issue as the OP reported. In the future, please open a new issue instead.

As to your problem: if you're building this from source, you'll need to make sure you have all prerequisites installed, such as in this case `hardware_interface` from `ros_control`. But that is not the only requirement, so it might make sense to follow the normal workflow and use `rosdep`. See [How to build a package from source on Ubuntu MATE 16.04 LTS and ROS Kinetic?](https://answers.ros.org/question/252478) on ROS Answers for an example workflow.