ros-industrial-attic/ur_modern_driver #65: Problem getting roslaunch to work with UR5 Polyscope version 3.3.0.145

p3jawors, 2016-09-13:
I know the main page says that the ur_modern_driver has only been tested up to v 3.0.16471 for the UR5, but has anybody had any luck running it on a later version? Unfortunately the arm I have was updated before I got it.

miguelprada, 2016-09-14:
I've used it just fine with v3.2.18744 on an UR10.

Could you elaborate on your problem?

carlosjoserg, 2016-09-14:
And I'm also using it with Polyscope 3.3.xx UR10

p3jawors, 2016-09-14:
> 246941097>,

Thanks for the quick reply! I've actually made a post on the ROS forum that
should have all the details.
http://answers.ros.org/question/243601/cannot-get-roslaunch-to-work-on-ur5-ros_industrial/
Let me know if there is anything else I can provide to help.

Cheers,

Pawel

miguelprada, 2016-09-14:
You can find my reply in [your post to ROS answers](http://answers.ros.org/question/243601/cannot-get-roslaunch-to-work-on-ur5-ros_industrial/), @p3jawors .

p3jawors, 2016-09-14:
Thanks for the reply. Would you prefer to have the conversation through
email or on the forum? For now I replied on the forum. Let me know your
preference.

Cheers,

Pawel

gavanderhoorn, 2016-09-15:
This turned out to be user error. See [answer](http://answers.ros.org/question/243601/cant-establish-connection-to-ur5-using-ros-industrial-driver/?answer=243799#post-id-243799) by @p3jawors.

closed 2016-10-24 by ThomasTimm