ros-industrial-attic/ur_modern_driver #66: Trajectory replacement during execution

harshmdeshpande, 2016-10-13:
I am currently working on trajectory replacement while the controller is still executing the current trajectory. The basic idea to send and execute the new trajectory when the target moves, without stopping the robot.  http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement

I have taken care to maintain continuity while merging the old trajectory with the new one. And I have tested this in Gazebo with fairly good results. The issue is that when I run this on real UR5, the robot stops before starting the new trajectory with the message :

`Received new goal while still executing previous trajectory. Canceling previous trajectory`

After going through the code, I found that `ur_ros_wrapper.cpp` stops the trajectory whenever a new trajectory arrives, in case there's already an existing goal in progress. Is there any decent way to not stop the robot but simply allow the remaining part of the trajectory to be replaced? I tried to simply comment out [this](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L289), but expectedly the robot did not behave nicely. 

Any suggestions are welcome. Thanks in advance!

ThomasTimm, 2016-10-24:
The iaea is really good, and I have some times considered if and how such a feature could be integrated.
The problem is that ROS action interface to the best of my knowledge doesn't support updating goals (see http://wiki.ros.org/actionlib/DetailedDescription ). So in order to achieve this, we would need to "abuse" the interface in a way that it is not intended to, and as such I have no Idea if the suggestion below would work.

The itnerface is meant for handling goals. A robot can only have one goal at a time (The arm can't go to two different poses at once), so when a new goal is received, any existing goals needs to be canceled. So in order to implement trajectory updating, we need a way to inform the driver that it is indeed an update and not a new goal. 
The intuitive way to do this would be to use the header ID, so when a client wants to update a goal, it sends a goal with the header ID of the goal existing goal that it wants to update. The problem is that I don't know how the action server reacts to getting a goal with the ID already set. Will it discard it as invalid (probably not, there isn't much input validation in ROS)? Will it overwrite the ID with a new number? Or will it leave it untouched?
If it is left untouched, it would "simply" be a matter of comparing the ID of a received goal to any currently active goal and, if they match, update the current goal instead of replacing it.

The easiest way to implement the update would probably be to add a method to the UrDriver class that sets the vectors timestamp, position and velocity, which should be private class variable, and then modify the doTraj method to use these variables instead of inputting them to the method.
To inform the client of the ID of the current running goal, the action interface feedback method should also be implemented.

harshmdeshpande, 2016-11-10:
Is it really necessary to make the driver believe that it's the same goal? As long as the user takes care that he maintains the continuity with existing trajectory, we could stop the current trajThread and start a new thread with the new goal?

Squelsh, 2016-11-10:
Just a comment:
The topic is in deed very interesting. I also need this feature for my reactive robot control.
But I am not sure, if you try to solve it at the best possible level...
Are you aware of the trajectory replacement in ROS-Control?
http://wiki.ros.org/joint_trajectory_controller/UnderstandingTrajectoryReplacement
http://wiki.ros.org/joint_trajectory_controller#Trajectory_replacement

harshmdeshpande, 2016-11-10:
Yes I am aware of this and have added this as a reference in my original post. But as I mentioned, 
https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L289, this doesn't allow the goal to be replaced, but stops the robot before accepting new goals. Any thoughts about Fanuc and ABB drivers? They seem to be using https://github.com/ros-controls/ros_controllers

Squelsh, 2016-11-10:
Uh, I am sorry, I thought you spoke about a UR-Modern specific function / problem.
Yeah, thats what I was using, too (the ros_controllers).
And then commanding the UR via position/speed control. But this required a dedicated network interface for me, otherwise I got ugly lags and jitter in the execution.

htugal, 2017-08-29:
I have a similar problem with my UR3 robot as .waitForResult() takes quite amount of time, I want to send new trajectory even though the previous one hasn't been finished.. 
So any update regarding to this issue ? 

Thanks,

geweihgg, 2018-05-31:
Hi, @harshmdeshpande ,
I'm using UR5 in gazebo to do some experiments. Now I want to control UR5 with joint speed commands, but I don't know how to configure the ur5.launch file so that I can control the arm in gazebo with ur_modern_driver.
Could you please tell me how to do this?
Thank you very much.
GeWei

gavanderhoorn, 2018-05-31:
@geweihgg wrote:
> I don't know how to configure the ur5.launch file so that I can control the arm in gazebo with ur_modern_driver.

as I wrote in #176, this driver cannot be used with a robot in Gazebo.

Please see #176 for another approach.

gavanderhoorn, 2019-11-23:
Trajectory replacement is a feature the `JointTrajectoryController` of `ros_control` supports.

`ur_robot_driver` is based on `ros_control`, so this could cover the use-case reported in the OP.

closed 2019-11-23 by gavanderhoorn