ros-industrial-attic/ur_modern_driver #70: Migrate RobotHW to combiner_robot_hw

paulbovbel, 2016-10-28:
Thanks for developing this fantastic driver.

I'm working on moving the Clearpath Husky to a combined_robot_hw implementation for kinetic. This has the advantage of using only one controller_manage per hardware robot, with multiple HW endpoints (base and arm) as well as controllers loaded.

Would you consider adapting your driver to do the same? It's kind of lacking documentation, but exciting nonetheless (see video here https://github.com/ros-controls/ros_control/issues/237)

gavanderhoorn, 2017-11-24:
Has the "kind of lacking documentation" situation been improved?

gavanderhoorn, 2019-11-23:
Note that initial support for `CombinedRobotHW` was merged in `ur_robot_driver` in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/34.

closed 2019-11-23 by gavanderhoorn