ros-industrial-attic/ur_modern_driver #73: No support for teach_mode

NikolasE, 2016-11-06:
I'd like to be able to activate the freedrive/teachmode. I had a look at the code and hoped that a simple 

```
void UrDriver::activateTeachMode() {
    rt_interface_->addCommandToQueue("teach_mode()\n");
    print_warning("Starting teach mode");
}
```
(with a service interface in the wrapper would work). However, the robot stays fixed after I called this function. This my first contact with the driver, so I'm not sure what to expect. 

Could maybe someone explain the difference between:

```
void UrDriver::stopTraj() {
	executing_traj_ = false;
	rt_interface_->addCommandToQueue("stopj(10)\n");
}
```
and
```
void UrDriver::setToolVoltage(unsigned int v) {
	char buf[256];
	sprintf(buf, "sec setOut():\n\tset_tool_voltage(%d)\nend\n", v);
	rt_interface_->addCommandToQueue(buf);
	print_debug(buf);
}
```

fjonath1, 2016-12-16:
To enable teach mode, what you need to do is simply use ```robot_.rt_interface_->addCommandToQueue('set robotmode freedrive')```
To disable teachmode, replace `set robotmode freedrive` with `set robotmode run`

If you're using ros, you could do that easily by sending that message to /ur_driver/URScript.
The code in python: 
```
ur_script_pub = rospy.Publisher('/ur_driver/URScript', String, queue_size=10)
if enable_teach_mode:
   ur_script_pub.publish('set robotmode freedrive')
else:
   ur_script_pub.publish('set robotmode run')
```

Hope that helps

NikolasE, 2016-12-17:
Could you explain the syntax? I can turn off the robot by .publish(shutdown()) (or something like that), 
so why can't i use publish(teach_mode()) ?

fjonath1, 2016-12-17:
As far as I know, the script required to be received by for my universal robot to enable/disable teachmode (with URScript) is `set robotmode freedrive` and `set robotmode run`. You might need to check what string you need to use for enabling teachmode in your robot.

Note: Before using this package, I used [python_urx](https://github.com/SintefRaufossManufacturing/python-urx) , which enable/disable the teachmode by sending `set robotmode freedrive` and `set robotmode run`.

```
# make a ros publisher that communicates with rosnode that will publish the received string to robot with rt_interface->addCommandToQueue
ur_script_pub = rospy.Publisher('/ur_driver/URScript', String, queue_size=10)

if enable_teach_mode:
   # send URScript command that will enable teach mode
   ur_script_pub.publish('set robotmode freedrive')
else:
   # send URScript command that will disable teach mode
   ur_script_pub.publish('set robotmode run')
```

ycheng517, 2017-01-05:
the freedrive command is not working for me. I've tried:
```
$ echo set robotmode backdrive | telnet 192.168.128.58 29999
```
for ports 29999, 30001, 30002. I've also tried: 
```
$ rostopic pub /ur_driver/URScript std_msgs/String "set robotmode freedrive"
```
Maybe a helpful piece of information, but these commands work:
```
$ rostopic pub /ur_driver/URScript std_msgs/String "power on"   # and for "power off" too
$ echo power on | telnet 192.168.128.58 29999   # and for "power off" too
```

happygaoxiao, 2017-12-09:
@fjonath1  I have tried your recommendation. But the teachpad doesn't receive any _log_ . 
@ycheng517  Have you solved this problem?  Hope someone could give me some advice. Thanks a lot.

happygaoxiao, 2017-12-09:
I publish _'freedrive_mode()'_ to  the rosnode "_ur_driver/URScript_" . The teach pad **log interface** receives message and start freedrive mode, but stop freedrive mode after 32ms. I try to add `sync()` after the String, but The teachpad shows `sync start` and `sync stop`.  I have also tried the `Thread():` or `def MyFun():`, still in vain.
     I write a simple **Script Code** in the teach pad, as below:
```
freedrive_mode()
sync()
```
and then run the script, it works as the freedrive mode. 
So, how to enable the freedrive mode or force mode with ROS node "_ur_driver/URScript_" ?

happygaoxiao, 2017-12-09:
I have solved this problem. The code is as below. We need a loop the same as  the teach pad script **Program loop forever** selected. The io_button refers to the _io_state_ of tool digital input.
```
        std_msgs::String temp;
        std::string cmd_str;
//        std::string force_mode="force_mode( tool_pose(), [0, 0, 1, 0, 0, 0], [0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 2, [0.05, 0.05, 0.05, 0.17, 0.17, 0.17])\n";
//        std::string free_drive_mode = "\tfreedrive_mode()\n";

        while(ros::ok()){
            bool last_io_button = io_button_;
            ros::spinOnce();
            if(!last_io_button && io_button_){
                cmd_str = "def myProg():\n";
                cmd_str += "\twhile (True):\n";
                cmd_str += "\t\tfreedrive_mode()\n";
                cmd_str +="\t\tsync()\n";
                cmd_str += "\tend\n";
                cmd_str +="end\n";
                temp.data = cmd_str;
                pub_free_drive_.publish(temp);
            }
            if(last_io_button && !io_button_){
                cmd_str = "def myProg():\n";
                cmd_str += "\twhile (True):\n";
                cmd_str += "\t\tend_freedrive_mode()\n";
                cmd_str +="\t\tsleep(0.5)\n";
                cmd_str += "\tend\n";
                cmd_str +="end\n";
                temp.data = cmd_str;
                pub_free_drive_.publish(temp);
            }
```
I will try  the force mode today.

ri-ceres, 2019-06-21:
I haven't had any success with the methods here. Enabling freedrive mode in [python-urx](https://github.com/SintefRaufossManufacturing/python-urx) works as expected, but not when using ROS. I've noted that the command in python-urx [has changed](https://github.com/SintefManufacturing/python-urx/commit/73bafadadfb43bc8a099b048f72122d547f2ef1d#diff-fd6d0c57706886dd0d20d8c29e06e3db) from `set robotmode freedrive` to `def myProg():\n\tfreedrive_mode()\n\tsleep(60)\nend`, but publishing to the `/ur_driver/URScript` topic doesn't have an effect on the robot.

This issue has been open for a few years now -- has anyone been able to get freedrive mode enabled recently?

HW: UR5e
SW: 5.2.0.61336

ri-ceres, 2019-06-21:
I see that if I set `shutdown_on_disconnect:=false` I can avoid totally killing the hardware interface, but then I get the unresolved issue noted in https://github.com/ros-industrial/ur_modern_driver/issues/291#issuecomment-483538645:

```
[ERROR] [1561148884.147129710]: Program upload failed!
```
@miguelprada suggested in #291 `stopping and starting the controller`, but it's not immediately clear how to properly do that from rospy.

happygaoxiao, 2019-06-22:
@ri-ceres  Hello, I made it at UR5 and 3.+software, with the `roslaunch` file `UR5_bringup.launch`. Are you using the `UR5e_roscontrol.launch` file? Actually in `roscontrol` mode, the robot is completely controlled by our PC, so the UR hardware cannot run `script` program, such as `teachmode `or `forcemode`. 

We also want to send UR script under the `roscontrol` mode. If you have any ideas, please tell me. Thanks a lot.

ri-ceres, 2019-06-25:
@happygaoxiao Similar to what you have described, we have only been able to enable teach mode with `UR5_bringup.launch` and **not** while using `ros_control`.

gavanderhoorn, 2019-11-23:
As `ur_robot_driver` contributes an *External Control* mode to urscript, it is fully integrated with the TP and the regular workflow there.

If External Control is not active, teach mode should be usable as normal.

If that is not the case, please open an issue on the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn