ros-industrial-attic/ur_modern_driver #75: Support for UR3 version 3.3.2.266?

mat-l, 2016-11-09:
Hello all,
we tried to control a real UR3 with this "ur_modern_driver".
The resulting Error message was more or less always the same. ROS could not load the robot controller. There war only one exception with the ur3_ros_control.launch. With this the robot reacted when pressing against it resulting in oversteering. could be a problem caused by the PID controller.

Furthermore,  the start, using the following command "~/catkin_ws_ur3/src/universal_robot-indigo-devel/ur_modern_driver/launch$ roslaunch ur3_bringup_joint_limited.launch robot_ip:=172.31.1.150 [reverse_port:=REVERSE_PORT]" is not working.
[image]

gavanderhoorn, 2016-11-11:
I've closed the duplicate over at ros-industrial/universal_robot#270.

@ThomasTimm: could you comment on compatibility with the binary protocol(s) used in `v3.3.2.266`?

ThomasTimm, 2016-11-28:
I (still) don't have access to a robot with firmware version higher than 1.8, so I am always working blind when trying to debug these issues... But please tell me you didn't literally type in `roslaunchur3_bringup_joint_limited.launch robot_ip:=172.31.1.150 [reverse_port:=REVERSE_PORT]`, because then I can easily tell you what the problem is, if you haven't figured it out yet?

gavanderhoorn, 2016-11-28:
@ThomasTimm: using ursim is not an option?

ThomasTimm, 2016-11-28:
It is kind of difficult to probe a simulated robot to see the oversteering. Generally, working with ros_control-related issues and the simulated robot isn't very optimal.
But to answer your question regarding the binary protocol for v3.3.x, I have no idea about compatibility. https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/remote-control-via-tcpip-16496/ still haven't been updated with information regarding firmware 3.3

gavanderhoorn, 2016-11-28:
re: dynamic behaviour: I completely understand and agree.

re: protocol changes: hm, that's not too nice. Perhaps UR wants to encourage users to migrate to RTDE?

ThomasTimm, 2016-11-28:
I have no idea if they changes the protocol yet again. Hopefully I will get a new UR in a month or two, but until then, I have no real way of figuring out if anything has changed (But a lot of other things to do though)

gavanderhoorn, 2016-11-28:
Yeah. RTDE has been around for some time now, as has v3.3.x; not sure what is going on.

See [Real-Time Data Exchange (RTDE) Guide - 22229](https://www.universal-robots.com/how-tos-and-faqs/how-to/ur-how-tos/real-time-data-exchange-rtde-guide-22229/) for some info on RTDE btw.

jettan, 2016-12-05:
Hi, I currently have access to a UR5 CB3 running v3.3.1.237
I don't see any issues with the reverse port connection and I have confirmed the robot moving fine. The ros_control driver is loading too, but has different issues (might be on my side).

james2254, 2016-12-28:
@ThomasTimm @jettan Hi I don't have access to a UR5 CB3 running v3.2.18744, but succeed 3days ago. This problem can be solved now?
[image]

gavanderhoorn, 2017-11-24:
I'm going to close this as I have multiple confirmations that `3.3.x` is working.

If this is still a problem @mat-l, please update and re-open.

closed 2017-11-24 by gavanderhoorn