ros-industrial-attic/ur_modern_driver #77: Several collision in a "normal" state

Raffa87, 2016-11-15:
Hi!
I tried to send a goal to the arm using moveit, but often a solution is not found. Reason is reported belove and it seems to be related to several self-collision, even if these are not present in the reality

[ INFO] [1479220669.062972752]: ParallelPlan::solve(): Solution found by one or more threads in 0.005977 seconds
[ INFO] [1479220669.077722763]: SimpleSetup: Path simplification took 0.014617 seconds and changed from 3 to 25 states
[ERROR] [1479220669.103667189]: Computed path is not valid. Invalid states at index locations: [ 66 67 92 93 97 98 120 121 122 123 124 125 ] out of 128. Explanations follow in command line. Contacts are published on /move_group/display_contacts
[ INFO] [1479220669.103899674]: Found a contact between 'forearm_link' (type 'Robot link') and 'wrist_3_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.103957870]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.104432579]: Found a contact between 'forearm_link' (type 'Robot link') and 'wrist_3_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.104489440]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.104913423]: Found a contact between 'wrist_1_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.104969817]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.105618323]: Found a contact between 'wrist_1_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.105675486]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.106119876]: Found a contact between 'wrist_1_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.106177171]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.106761201]: Found a contact between 'wrist_1_link' (type 'Robot link') and 'upper_arm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.106816520]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.107394690]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.107450645]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.108076428]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.108131404]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.108724000]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.108782007]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.109281091]: Found a contact between 'wrist_2_link' (type 'Robot link') and 'forearm_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.109335704]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.109676050]: Found a contact between 'forearm_link' (type 'Robot link') and 'wrist_2_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.109728354]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1479220669.110080487]: Found a contact between 'forearm_link' (type 'Robot link') and 'wrist_2_link' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[ INFO] [1479220669.110132711]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ERROR] [1479220669.110358637]: Completed listing of explanations for invalid states.

gavanderhoorn, 2016-11-16:
Also [Several collisions in a "normal" state #7](https://groups.google.com/d/msg/moveit-users/Fr6E6YQS7RA/dgXUi12DAQAJ) in `moveit-users`.

Raffa87, 2016-11-16:
Yes, I am the author of both.
I cannot understand if it is a problem related to driver or to moveit!

ThomasTimm, 2016-11-16:
Its a problem related to moveit. Collisionchecking is done at the planning level, not in the driver

closed 2016-11-16 by ThomasTimm