ros-industrial-attic/ur_modern_driver #78: Corrections needed for default ros_control configuration

HurchelYoung, 2016-11-18:
I just finished setting up the ur_modern_driver using ros_control with the velocity based position control.  I was following the suggested usage instructions in the readme but I ran into some issues. 

Primarily it was the suggested naming of the controllers.  
The suggested names are:
- vel_based_pos_traj_controller #or pos_based_pos_traj_controller 

I believe the names need to be:
- /velocity_based_position_trajectory_controller #or /position_based_position_trajectory_controller

It was also not entirely clear to me that when implementing the ros_contol based approach that I would need packages that are found in the old universal_robots driver, namely the urXX_movit_config and ur_description packages.  The readme just needs a couple of sentences to point this out.

If you agree with these corrections, I've made them to my branch and can submit a pull request.

HurchelYoung, 2016-11-21:
@miguelprada pointed out that these changes made the ur10 and ur5 config files disagree with the documentation.  It seems that the ur3 was the original odd man out.
Since nobody has mentioned this before I'm guessing the majority of users are working with the ur10 and the ur5.  I feel like it would be best to update the ur3 to coincide with those files.

miguelprada, 2016-11-21:
I think I agree, although your latest commits in #79 have the opposite changes (i.e. modify UR5 and UR10 configurations to match UR3).

HurchelYoung, 2016-11-21:
Yeah, sorry about that.  I made those changes, then thought better of it, and then immediately got distracted by a squirrel.  Anyway, my latest commits should bring everything into agreement.

ThomasTimm, 2016-11-28:
Thanks for the commit! I've merged the changes

closed 2016-11-28 by ThomasTimm