ros-industrial-attic/ur_modern_driver #81: Publishing Joint Temperature and information

atomoclast, 2016-11-22:
Hello all,

I'm in the process of developing a system with several UR robots and we are using the ur_modern_driver package to allow for us to move the robots with ros_control. 

One of the things that has come up recently for our application was monitoring the health of the robots, and we were interested in logging the temperature of the joints of each robot with respect to time. On the UR pendant, on the "Log" tab, we see the temperature being displayed. Is there any way to use the ur_modern_driver package for us to be able to get that information published via ros? Or is there a way to get that data and log it?

Thanks in advance!

miguelprada, 2016-11-23:
There should be a way, since that information is sent by the controller through the RT interface and unpacked by `RobotStateRT` (see the [unpacking](https://github.com/ThomasTimm/ur_modern_driver/blob/af6edc8a3552c449d27076db1642adf8afa97714/src/robot_state_RT.cpp#L394) and the [getter](https://github.com/ThomasTimm/ur_modern_driver/blob/af6edc8a3552c449d27076db1642adf8afa97714/src/robot_state_RT.cpp#L228)).

It should be a matter of reading these values in [`rosControlLoop`](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L565) and publishing them as desired (I suggest using a `realtime_tools::RealtimePublisher`, such as [this one](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_ros_wrapper.cpp#L576)).

ThomasTimm, 2016-11-28:
Do any of you have any suggestions as to what message type we should use for this? - preferably something that is already in ROS

gavanderhoorn, 2016-11-30:
`sensor_msgs` contains a `Temperature` msg: [sensor_msgs/Temperature](http://docs.ros.org/kinetic/api/sensor_msgs/html/msg/Temperature.html).

I would suggest to use that.

Perhaps a single publisher would suffice, if we set the `frame_id` to suitable values.

ThomasTimm, 2016-12-01:
So should we set up 1 (or six) publishers that just publishes joint temperatures, or a more elaborate system state message overview?

gavanderhoorn, 2016-12-01:
A single publisher should be possible:

> frame_id is the location of the temperature reading

if the TF frames for the joints are acceptable as 'locations' of the motors.

gavanderhoorn, 2017-11-24:
Probably covered by #120.

gavanderhoorn, 2018-10-05:
This was added in https://github.com/ros-industrial/ur_modern_driver/pull/120: motor temperatures for all joints are published.

closed 2018-10-05 by gavanderhoorn