ros-industrial-attic/ur_modern_driver #82: Fails to compile under Jade due to changed controller_manager_msgs structure

Josque, 2016-11-24:
To quote [this website](https://github.com/ros-planning/moveit_ros/issues/656). I think I have the same problem.

> I did a little more investigating and it would seem that the structure of controller_manager_msgs has changed between Indigo and Jade, such that ControllerState.msg has gone from this:
> string name
> string state
> string type
> string hardware_interface
> string[] resources
> 
> to this:
> 
> string name
> string state
> string type
> controller_manager_msgs/HardwareInterfaceResources[] claimed_resources
> 
> where HardwareInterfaceResources.msg now contains the resources field:
> 
> /# Type of hardware interface
> string hardware_interface
> /# List of resources belonging to the hardware interface
> string[] resources
> 
> I will revert back to Indigo for now and will try building MoveIt from there, but I hope the above might be helpful for developing the Jade release. I would try to fix it myself, but I'm not familiar enough with the codebase and don't have the time right now.
> 
> Cheers,
> Barry
>
Can you change this?

I hope that fixes this error:
```
/home/user/Projects/ROS/Training/src/ur_modern_driver/src/ur_hardware_interface.cpp:180:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
        if (controller_it->hardware_interface == "hardware_interface::VelocityJointInterface")
```

Josque, 2016-12-02:
Here is the error message:
`ur_modern_driver/src/ur_hardware_interface.cpp:180:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
   if (controller_it->hardware_interface == "hardware_interface::VelocityJointInterface") {
                      ^
In file included from /opt/ros/jade/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /opt/ros/jade/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /opt/ros/jade/include/hardware_interface/joint_state_interface.h:33,
                 from /home/user/Projects/ROS/Training/src/ur_modern_driver/include/ur_modern_driver/ur_hardware_interface.h:61,
                 from /home/user/Projects/ROS/Training/src/ur_modern_driver/src/ur_hardware_interface.cpp:58:
/home/user/Projects/ROS/Training/src/ur_modern_driver/src/ur_hardware_interface.cpp:185:22: error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’
       controller_it->hardware_interface.c_str());
                      ^
`

Josque, 2016-12-02:
the problem is that hardware_interface.h defers in jade and indigo.
this is the file in **Indigo**:

> #ifndef HARDWARE_INTERFACE_CONTROLLER_INFO_H
> #define HARDWARE_INTERFACE_CONTROLLER_INFO_H
> 
> #include <set>
> #include <string>
> 
> namespace hardware_interface
> {
> 
> /** \brief Controller Information
>  * This struct contains information about a given controller.
>  */
> struct ControllerInfo
> {
>   std::string name, type, hardware_interface;
>   std::set<std::string> resources;
> };
> 
> }
> 
> #endif
> 

and this is the code in **Jade**:

> #ifndef HARDWARE_INTERFACE_CONTROLLER_INFO_H
> #define HARDWARE_INTERFACE_CONTROLLER_INFO_H
> 
> #include <string>
> #include <vector>
> 
> #include <hardware_interface/interface_resources.h>
> 
> namespace hardware_interface
> {
> 
> /** \brief Controller Information
>  *
>  * This struct contains information about a given controller.
>  *
>  */
> struct ControllerInfo
> {
>   /** Controller name. */
>   std::string name;
> 
>   /** Controller type. */
>   std::string type;
> 
>   /** Claimed resources, grouped by the hardware interface they belong to. */
>   std::vector<InterfaceResources> claimed_resources;
> };
> 
> }
> 
> #endif

As you can see the indigo headerfile uses a string for the name, type and hardware_interface where the Jade headerfile uses a vector.

daveying, 2016-12-12:
Same problem!!

Josque, 2016-12-22:
@ThomasTimm Do you have a solution to this?

ThomasTimm, 2016-12-22:
At the moment, nothing except suggesting not to use Jade... If that is not an option, I'll be happy to see pull requests from the community to solve the problem.

BenBlumer, 2017-01-27:
See solution here: https://github.com/ThomasTimm/ur_modern_driver/issues/58

gavanderhoorn, 2017-11-24:
As Jade is EOL, closing this.

The problems with Kinetic are known, and tracked in issues such as #58 mentioned by @BenBlumer.

gavanderhoorn, 2017-11-24:
Not ignoring the build issues @Josque, but tracking it in a different issue.

closed 2017-11-24 by gavanderhoorn