ros-industrial-attic/ur_modern_driver #83: Are the tool analog inputs available?

NikolasE, 2016-11-25:
I haven't found out how to read the analog inputs of the tool, did I miss it?

ThomasTimm, 2016-11-28:
It should be published as part of the i/o messages

NikolasE, 2016-11-28:
The io-messages are parsed from the masterboarddata (or something like that) and the tool inputs are not in this data block. I succeeded with the rtde-Interface, an example is in my fork: https://github.com/NikolasE/ur_modern_driver

ThomasTimm, 2016-11-28:
Ahh yea, I never noticed that. 
It can also be parsed from the Tool data block, that way we won't have to set up another connection.

NikolasE, 2016-11-28:
How much work is it to get and parse the tool data block? And how fast is it?

ThomasTimm, 2016-11-28:
a rough estimate is 100-200 lines of code injected the right places :)
The data rate is 10 hz afair.

NikolasE, 2016-11-28:
If you point me to the right places, I could be tempted to fill them. The modern_driver works well so if I'd love to contribute.

NikolasE, 2016-11-29:
I started to implement the ToolData: https://github.com/NikolasE/ur_modern_driver/commit/13dc20e8549e1d149a8e8529598005e710ecc7ed

I wrote some parsing functions to have clearer code and to simplify the implementation of other data-packages. 

So far, I did not succeed in parsing the floats, and I saw that you read floats like this:

```
	memcpy(&mb_data_.masterBoardTemperature, &buf[offset],
			sizeof(mb_data_.masterBoardTemperature));
	offset += sizeof(mb_data_.masterBoardTemperature);
	mb_data_.masterBoardTemperature = ntohl(mb_data_.masterBoardTemperature);
```
ntohl returns an uint32_t and in the current implementation, the value I get is always zero. Could you maybe explain your approach here a bit? 

What do you think about fixing the parseFloat-function and then rewriting the other functions with the parse* so that it's a bit more readable?

gavanderhoorn, 2016-12-30:
I just wanted to mention https://github.com/ros-industrial/universal_robot/pull/272#issuecomment-269660687 here as a point of discussion.

@ThomasTimm: what is your opinion on this? Is `ur_modern_driver` intended as a transparent bridge (dumping raw UR protocol msgs as ROS msgs on topics, delegating responsibility of interpreting those messages to consumers) or as a proxy / facade with a ROS API (and interpreting the data itself, making sure ROS msgs are stand-alone and unambiguously interpretable without needing robot configuration data)?

With the addition of the `ToolData` msg in ros-industrial/universal_robot#272, we have taken a step in the direction of a transparent bridge. I think we need to carefully consider this, as transparent bridges make proper abstraction more difficult.

closed 2019-11-23 by gavanderhoorn