ros-industrial-attic/ur_modern_driver #85: Why not Boost::Asio?

facontidavide, 2016-12-05:
Hi,

a quick question.
Boost is a well know dependency of ROS. Is there any reason why Boost::Asio can't be use to make the part of the code related to communication more "terse"?

Davide

ThomasTimm, 2016-12-05:
There are a lot of reasons, some better than others... a few quick ones are that the author has zero experience with it, and it wouldn't add any real value to the project to use time to learn it. Also a desire to keep the code reusable by having as few dependencies as possible (see #84 as well). Except for the two ROS-related files that exposes the driver to ROS, the driver doesn't require anything else than c++11, making it easily portable to other systems without ROS.

closed 2016-12-05 by ThomasTimm

facontidavide, 2016-12-05:
Thanks for your reply.

I quite frankly take with a grain of salt the importance of "no dependency" requirement, especially when it is a apt-get away...

The advantage of boost is that it would make the code portable to Windows.
Anyway, I agree with you that it ill not add any value to the project, except portability to other platforms.

Feel free to close this issue ;)