ros-industrial-attic/ur_modern_driver #92: Can we really omit time argument in speedj?

sergeant-wizard, 2017-01-17:
Regarding [this call](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_realtime_communication.cpp#L103)
to `speedj`, I was having an issue where the published `/joint_states/velocity` was oscillating like this:

[image]

Where I was publishing to `/ur_modern_driver/joint_speed` at 125Hz.
When I added the `time` argument in `speedj` as `0.008` in the above line, I got a smooth output:

[image]

I noticed that @ThomasTimm mentioned in #55 that the time argument doesn't really matter, but just wanted you to know that I needed to specify the argument to get stable results.

The UR software version is 3.3.1.237, and I'm using UR5.
ROS distro is Indigo on Ubuntu 14.04.

mitch1044444, 2017-06-28:
@sergeant-wizard Hey, I wonder if you could share how to plot this? I couldn't find much information on how to use URscript in ROS, and I'm afraid I might break our robot here, any help would be deeply appreciated.

sergeant-wizard, 2017-06-28:
@mitch1044444 
I published the joint velocities to `ur_driver/joint_speed` as `JointTrajectoryPoint`s, subscribed to '/joint_states' and recorded the latest of each at every control frame.
Don't worry, your robot will not break if you start with very small velocities.

sergeant-wizard, 2017-06-28:
https://gist.github.com/sergeant-wizard/bd9d305d4f02e0caa424cde4af0537ce

something like this, though it hasn't been tested

gavanderhoorn, 2017-11-24:
Related to #55 and #62.

inksci, 2018-05-14:
You have mentioned that:
"I published the joint velocities to ur_driver/joint_speed as JointTrajectoryPoints"

Therefore, could you tell me that if I use the speedj for /ur_driver/URScript, rather than use ur_driver/joint_speed, will the results be same?

gavanderhoorn, 2019-11-23:
This should no longer be an issue as `ur_robot_driver` uses a somewhat different approach to controlling the robot.

If this *is* still an issue, please report it over at the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn