ros-industrial-attic/ur_modern_driver #95: ur_modern_driver sometimes keeps executing one trajectory list behind

fjonath1, 2017-02-28:
Hi,

My research group at university has been using ur_modern_driver for awhile, and sometimes we noticed a strange bug where ur_modern_driver never finishes executing a trajectory (it just stopped halfway). Whenever this happened, after we cancel the goal, if we input a new trajectory list, it will be executing 1 trajectory list behind.

here is the log we have when it happened. Note: sec 24 in `follow_joint_trajectory/result` and sec 26 in `/follow_joint_trajectory/goal` is the last published msg/response we got before cancelling the goal.
Afterward, the `follow_joint_trajectory/result` always says error code: 0 and error_string: Goal canceled by client. The published joint state reflects the actual robot joint position.

Any helps would be appreciated,
Felix

`rostopic echo /follow_joint_trajectory/goal`
```
---
header: 
  seq: 25
  stamp: 
    secs: 1488322862
    nsecs: 751332044
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1488322862
    nsecs: 751256942
  id: /simple_ur5_driver_node-25-1488322862.751
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: ''
    joint_names: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
    points: 
      - 
        positions: [0.30003762358163316, -1.3300827834477285, -1.7997786308312367, -0.2700036285261235, 1.4998983234390535, 1.6002492253792502]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [0.2740209366499169, -1.3557743539621965, -1.8373377133488311, -0.1987249852305596, 1.5041957780008162, 1.6021728750855273]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 169328642
      - 
        positions: [0.22599013000674842, -1.4032049456812143, -1.906677557996698, -0.06713364376182626, 1.5121295402686858, 1.6057242283894235]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 338657284
      - 
        positions: [0.16502795234426537, -1.463405312093814, -1.994685822357452, 0.09988613579464298, 1.5221993154548281, 1.6102317152751378]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 507985927
      - 
        positions: [0.10250264192120583, -1.525149277645198, -2.084950708881302, 0.2711884738012781, 1.5325272900047175, 1.6148547787476653]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
[... 356 lines cut ...]
      secs: 1488323109
      nsecs: 282537937
    id: /simple_ur5_driver_node-28-1488323109.283
  status: 3
  text: ''
result: 
  error_code: 0
  error_string: Goal cancelled by client
---
```

The last `/joint_states` when the robot never finishes/execute the trajectory we sent.
```
header: 
  seq: 93739
  stamp: 
    secs: 1488322875
    nsecs: 735777683
  frame_id: ''
name: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
position: [-0.038390898830618525, -1.66428467107084, -2.28834839267841, 0.6571372359277533, 1.5557814352918464, 1.6252728742097837]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [-0.2981608185299103, 1.7777558578512693, 1.4616605539962517, 0.9348502409565058, 0.19215518819008112, 0.4468370646007442]
---
```

xqms, 2017-03-24:
This may be related to my PR #101 which addresses some race conditions present in the trajectory action goal/cancel callbacks. Maybe you can check if you can still reproduce the problem with #101?

I also noticed that the error string is never reset, so subsequent action calls after a failure will always return the old error string.

fjonath1, 2017-03-24:
> This may be related to my PR #101 which addresses some race conditions present in the trajectory action goal/cancel callbacks. Maybe you can check if you can still reproduce the problem with #101?

Interesting. I do noticed that this problems seems to happen more often with more stop trajectory in the middle of execution, although it usually happened randomly. That might be in fact related to the issue you addressed on your PR.

I will try #101 and get back to you. It might take few days to ascertain whether it fixes this problem or not, given the randomity behavior of this bug.

a-price, 2017-07-25:
I'm seeing this behavior on 9ca88362427e32f8d1a125199f6b8e387e81a711. I'll try #101 as well and report back.

a-price, 2017-07-25:
I haven't seen the retained trajectory bug on #101, but the server appears to be deadlocking after running for a period of time.

Output from the follow\_joint\_trajectory/status channel:

    ---
    header: 
      seq: 2614
      stamp: 
        secs: 1500952834
        nsecs: 250381600
      frame_id: ''
    status_list: 
      - 
        goal_id: 
          stamp: 
            secs: 1500952750
            nsecs: 823723015
          id: /capture_experiment-96-1500952750.823723015
        status: 2
        text: ''
      - 
        goal_id: 
          stamp: 
            secs: 1500952781
            nsecs:  77816914
          id: /capture_experiment-97-1500952781.77816914
        status: 7
        text: ''

Output from node:

    [ INFO] [1500952750.868821747]: on_goal
    [DEBUG] [1500952750.869189614]: Accepting goal, id: /capture_experiment-96-1500952750.823723015, stamp: 1500952750.82
    [DEBUG] [1500952763.989070505]: The action server has received a new cancel request
    [DEBUG] [1500952763.989228775]: Transisitoning to a cancel requested state on goal id: /capture_experiment-96-1500952750.823723015, stamp: 1500952750.82
    [ INFO] [1500952763.989467065]: on_cancel
    [DEBUG] [1500952763.989660130]: Setting status to canceled on goal, id: /capture_experiment-96-1500952750.823723015, stamp: 1500952750.82
    [DEBUG] [1500952763.989778564]: Publishing result for goal with id: /capture_experiment-96-1500952750.823723015 and stamp: 1500952750.82
    [DEBUG] [1500952781.141009364]: The action server has received a new goal request
    [ INFO] [1500952781.141109694]: on_goal
    [DEBUG] [1500952812.011810600]: The action server has received a new cancel request
    [DEBUG] [1500952812.011858699]: Transisitoning to a cancel requested state on goal id: /capture_experiment-96-1500952750.823723015, stamp: 1500952750.82
    [DEBUG] [1500952812.011892001]: Transisitoning to a cancel requested state on goal id: /capture_experiment-97-1500952781.77816914, stamp: 1500952781.08
    
Action 96 did not execute anything (maybe the deadlock starts here?), so the client timed out and sent a cancel request. Notice that the debug statements say it's trying to cancel the 96 goal twice. After this point, the node is completely locked, and sending additional goals or cancels has no effect and produces no debug statements.

This behavior has been observed multiple times over the past hour or so.

danielkruse, 2017-12-19:
I am also observing the retained trajectory bug issue.  Ours seems to be triggered infrequently by a trajectory cancellation.  Has [#101](https://github.com/ThomasTimm/ur_modern_driver/pull/101) demonstrated an appropriate fix?

smurray47, 2018-02-01:
I can't guarantee correctness, but we were having the same problem, and #101 has fixed it, or at least made it much more infrequent. I can also confirm we also experience very occasional deadlock with that pull request, but that is a MUCH more preferable mode of failure. 

This is a dangerous bug! I'm surprised that pull request has been sitting for a year. Our UR5 would freak out every once in a while when we cancelled and then gave a new trajectory.

prajval10, 2018-08-02:
I am also facing the same issue.  If I lower the velocity slider bar in the UR teach pendant and run the trajectory, the robot goes into an abrupt protective stop. After which, I enable the robot from the pendant and cancel the goals and send a new trajectory.  This time the robot goes into various jerky random motions before going to desired pose. This must be due to the retained trajectory. Has anyone found a proper fix to the problem? I will check #101.

gavanderhoorn, 2019-11-23:
`ur_robot_driver` makes use of `ros_control`'s `JointTrajectoryController`, which should not exhibit this behaviour.

closed 2019-11-23 by gavanderhoorn