ros-industrial-attic/ur_modern_driver #97: UR3 crashes when using ros_control approach

IlyaBoyur, 2017-03-15:
Hello Thomas and community,
the issue is when I start ros_control driver UR3 starts shaking and jiggling around it's current position. So I must interrupt driver to prevent robot from been damaged.

What do I have:
1) Notebook with Ubuntu Trustu 14.04.05 + ROS Indigo
2) Installed ROS pakages: ros_control, ur_modern_driver, moveit, universal_robot, ros_indigo_desktop

What am I doing::
1) Setting vel_based_controller in ur3_moveit_config/config/controllers.yaml;
2) starting roslaunch ur_modern_driver ur3_ros_control.launch robot_ip:=MY_IP
Here is the place when driver start to crash my robot.

What am I doing wrong? Is there additional robot relative configurations I can find , which are needed to be set before launching ros_control controller?

P.S: Please, refer to **bug in ur3_ros_control.launch file** in driver README which prevent driver from starting (or change this file):
In `

miguelprada, 2017-03-15:
Looks related to #55, #62 and #92.
Which software version is your controller running?

IlyaBoyur, 2017-03-31:
Thanks for reply, Miguel Prada.
Robot software version:  3.2.19293.
I am launching 
`roslaunch ur_modern_driver ur3_ros_control.launch robot_ip:=MY_IP` - here is there robot goes crasy

I am using source package for master branch and have built it by myself. (GCC 4.8.4)
Actually, I haven't found any binary source files for that driver and there are no instructions in `ur_modern_drive`r README how to get if any exists.

`pos_based_pos_traj_controller` seems to work for UR3 as mentioned in #55. Also I use `ur_modern_driver` with mentioned in #55 `robot_state_->getVersion() >= 3.1` condition and omitted`t` parameter in `speedj`. but I still have no clue how to vary PID for controller.

I tryed to use` t=0.008 `as  @sergeant-wizard describes in  #92. My robot simply goes unstable as it went earlier.
I haven't diagnosted much of the problem, since pos_based_pos_controller work. But it's preferably to use `speedj`, so I woundered if someone could give any hints what to do/check in my installation.

miguelprada, 2017-04-03:
If your robot goes unstable when using the velocity interface I suggest you try to adjust the PID parameters for `vel_based_pos_traj_controller` in [`ur3_controllers.yaml`](https://github.com/ThomasTimm/ur_modern_driver/blob/9ca88362427e32f8d1a125199f6b8e387e81a711/config/ur3_controllers.yaml#L79) since they have not been adjusted for a real robot.

mitch1044444, 2017-06-28:
@miguelprada Hey, I wonder what is the difference between 
`roslaunch ur_modern_driver ur3_ros_control.launch
roslaunch ur_modern_driver ur3_bringup.launch`

I was using the bringup method with moveit to control the robot, and I just find there is a new interface in ur_modern_driver, and I get really confused between this two. In my opinion, we can do "velocity control" in the bringup.launch right? 

I wonder if you could help me understand it.

Thankyou.

miguelprada, 2017-06-29:
> @miguelprada Hey, I wonder what is the difference between
roslaunch ur_modern_driver ur3_ros_control.launch roslaunch ur_modern_driver ur3_bringup.launch

This driver supports two _modes_ of operation: with or without `ros_control`.

The `urX_bringup.launch` files will start the driver in the non-`ros_control` mode, which AFAIK allows you to control it by using `JointTrajectory` messages/actions, by sending joint velocity commands, or by directly sending strings containing URScript code.

The `urX_ros_control.launch` files will start the driver in `ros_control` mode. Lets summarize a lot by saying that `ros_control` is a framework which allows you to implement controllers at the joint position/velocity level as plugins and to load/unload and start/stop them dynamically.

There's much more to it, though. This second _mode_ is probably the most versatile of both, but it also has a steeper learning curve. See this [ROSCon talk](https://vimeo.com/107507546) if you're interested in knowing more.

> In my opinion, we can do "velocity control" in the bringup.launch right?

According to the README, you can do joint velocity control by sending appropriately formated `JointTrajectory` messages to the `/joint_speed` topic, yes. I've never tried this, though.

If what you're after is controlling the tool's velocity you will need to either do it through URScript commands or by implementing a controller that does it using the `ros_control` _mode_.

mitch1044444, 2017-06-29:
@miguelprada Thank you for such a good explanation, I was thinking about the same thing. I saw this RosCon talk before and I wonder if you happen to have some ros_control example by hand? I googled some examples on github, or gazebo. But I don't find much joint_speed control example. It's ok if you don't have one at hand, I think I should try at least :).

Thank you again for your detailed explanation.

miguelprada, 2017-06-30:
If what you want is examples of `ros_control` controller plugins, you have plenty of those in the [`ros_controllers`](https://github.com/ros-controls/ros_controllers) package. They might not be the simplest possible examples, but they're good examples.

gavanderhoorn, 2017-11-24:
I believe this is indeed an instance of #55, #62 and/or #92 as @miguelprada suggested.

Closing this one in order to weed out some duplicates.

@IlyaBoyur: this doesn't mean that we're not tracking the issue. Just that we'd like to focus our attention and avoid disjoint discussions and potentially duplicated efforts.

closed 2017-11-24 by gavanderhoorn