ros-industrial-attic/ur_modern_driver #103: Joint states published in [-2pi, 2pi], despite limiting joints to [-pi, pi] in urdf

tanay-bits, 2017-04-14:
I'm integrating ur5 on a mobile base, and on my _wrapper xacro_, where I'm instantiating the base and the arm, I'm simply setting the `joint_limited` param to true as follows:
```
  <xacro:include filename="$(find ur_description)/urdf/ur5.urdf.xacro" />
  <xacro:ur5_robot prefix="${arm_name}" joint_limited="true" />  
```
Then I load the `robot_description` param with this wrapper xacro of my robot in the launch file (similar to _ur5_bringup.launch_). This should limit the joints from -pi to pi, but when I rostopic echo on joint_states, I still see the joints going from -2pi to 2pi. This causes MoveIt a lot of trouble planning, since it finds the start state outside the bounds [-pi, pi] which it expects. Could someone confirm what range are the joint states published by ur_modern_driver?

P.S. - I did try just making joint_limited to false to make MoveIt happy; it doesn't throw the out of bounds error, but it takes so long to plan that most often it just times out.

closed 2017-04-17 by tanay-bits

ThomasTimm, 2017-04-18:
I'll comment on it for future reference;

The driver tells ROS what the actual state of the robot is, and so it should be. If you require the robot to be limited to within a certain interval, and the robot is powered up outside that interval, you should have a way to detect that, and then move the arm into a valid state. If the driver "hides" the fact that the robot is outside its bounds, it would not be possible to do this recovery.

The joint limited xacro is only for helping the planner finding a solution. If you move your robot so all joints are within the [-pi;pi] interval and then exclusively use the  planner for moving the arm around, you shouldn't have any issues.

tanay-bits, 2017-04-18:
Thanks for your reply! I realized I emailed too soon; it was clear what I
was missing after reading a couple of issues on the universal robots github
page.