ros-industrial-attic/ur_modern_driver #105: U10 crashes when using ros_control

ane-fernandez, 2017-04-28:
Hi ThomasTimm and the community,

I'm trying to move a UR10 with ros_control approach and  don't get the expected behavior. I'm using a position control base controller. When a trajectory in sent to be executed, the robot start moving very abruptly and after a few seconds the driver crash saying that "Robot is protective stopped!".

At the beginning I was using Moveit to generate the trajectory and I thought it could be related with the way was generated, but I plotted them and I saw the path was quite smooth.
[image]

However, to avoid Moveit and try to isolate the problem, I execute the test_move.py file that was in ur_modern_driver and the robot behavior was the same. 

I have checked the servoj gain and lookahead param as is said in #47 issue, but they were ok.

What am I doing wrong? Is additional configuration need?

The Robot controller software version is:  3.1.17336

Thanks

fbe555, 2017-05-09:
How are you connected to the robot? I've experienced similar issues in cases where my connection wasn't super steady. As I recall, the driver reports 0 for all joint positions if the connection has a fallout, which then leads to an abrupt movement when reconnected.

ane-fernandez, 2017-05-17:
I'm using Ethernet cable to connect to the robot, and it is quite long. I can try with shorter cable to see if the problem remains. I will check also if the driver returns 0 for all joints positions. The joint state was shown automatically in case that the connection had a fallout or you had to add some extra ROS_INFO to the driver?

Thanks in advance

fbe555, 2017-05-17:
I wad having trouble due to a router. Don't think the long cable should be a problem. The joint states are published by the joint state controller (see ros_control), to the topic /joint_states

ane-fernandez, 2017-05-17:
I've being monitoring the joint_states topic and in none publication the joint position was 0, however the robot remains moving abruptly and after a few seconds the robot is stopped.

fbe555, 2017-05-17:
Then I don't think it's the same issue. 
I would try and replicate it, but we're using kinetic and waiting for the ur_modern_driver to be updated, so don't really have a working setup.

ane-fernandez, 2017-05-18:
Thank you very much @fbe555 for your help. After several test, I had the suspicion that the problem could be related with the Firmware version. In our facilities we have an UR10 with CB2 and I've run the same trajectory with velocity controller and it works well. I decided to update the Firmware of the UR10 to a newer version, and now both controllers work. I'm having some strange behaviours with velocity controller, but that is another issue.

gavanderhoorn, 2017-11-24:
@ane-fernandez: I have the impression (from your last comment) that this is no longer an issue, correct?

If so: could you please close this one?

closed 2017-11-24 by ane-fernandez