ros-industrial-attic/ur_modern_driver #108: How to use tool0_controller via MoveIt and how to use DH parameters

MingheJiang, 2017-05-11:
Hi, I am trying to get accurate move for ur5 MoveIt control based on the fact that each UR is calibrated individually and thus URDF is not accurate. This leads to the max error of 3 mm.

For now, I have used tool0_controller to read real end-effector pose successfully. However, it was not enough to just read it for getting accurate move target position. Is there any way to control that frame instead of ee-link or tool0 via MoveIt? 

From #16 , I also extracted the calibrated DH parameters from a UR robot. But I am not quite understand the meaning of some parameters, like joint_checksum, joint_raw_offset, joint_selftest_data_crc. Which parameter should I use to get an accurate ur5.urdf? @sepjansen 

Calibration.conf file:

[mounting]
delta_theta = [ -1.10635870277260494e-05, -0.774161735447248089, 0.727955647928133054, 0.0461047799534757691, -7.47409496122233342e-05, 4.83742328285148039e-05]
delta_a = [ 0.0002794266622372088, 0.12113393730165295, 0.00054449524997751686, 6.42184785383017725e-05, 0.000136602328054314255, 0]
delta_d = [ 0.000189349432381524796, 232.671212060261581, -229.362882562568046, -3.30684242648800275, 0.00024453960244240347, 0.000245740687010886916]
delta_alpha = [ 0.000109725903021828941, -0.00127681131702107572, -0.00546192611333388891, -0.000855429104392291251, 0.00050738191100685448, 0]
joint_checksum = [ 0xf2d99c79, 0x39d14081, 0x71a3b109, 0xa9499404, 0xd639e411, 0x48e237f6]
calibration_status = 2 # 0 == notInitialized / 1 == notLinearised / 2 == Linearised
joint_raw_offset = [ 0xd4ce47, 0xd26939, 0x531947, 0xfe458c, 0xfe4764, 0x7db9ea]
joint_selftest_data_crc = [ 0x8bcd8510, 0xa2d0ef15, 0xe192454b, 0x9e579e19, 0x6d337a6, 0x261722fb]

Thanks!

gavanderhoorn, 2017-11-24:
If you understand DH parameters, the values you show should allow you to update your urdf and get as close as possible to 'true' values with just the urdf. Beyond that, `tool0_controller` is available.

> For now, I have used `tool0_controller` to read real end-effector pose successfully. However, it was not enough to just read it for getting accurate move target position. Is there any way to control that frame instead of `ee-link` or `tool0` via MoveIt? 

I'm not sure I understand you here: the driver controls the robot in joint space, so at first glance, the answer would be: "no, you cannot control `tool0_controller`". However, it should be possible to write a Cartesian controller for `ros_control`. That controller could then close the loop over any arbitrary TF frame, including `tool0_controller`, and use either the position or velocity interfaces to implement a position control loop that applies the necessary corrections.

While this is possible, beware of some of the current issues, such as #139, #132 and #114.

gavanderhoorn, 2017-11-24:
As this is out-of-scope for this driver, I'm going to close the issue.

If you feel this is incorrect, please clarify and reopen @MingheJiang.

closed 2017-11-24 by gavanderhoorn