ros-industrial-attic/ur_modern_driver #109: UR5 I/O Support

sameergrey, 2017-05-11:
Hi Greetings!!!!
I am working on UR5 real robot CB2-firmware version >3.2.
I want following task to be done by my Robot using the ur_modern_driver.

- I want to connect a analog sensor to the analog pin of UR5 and to monitor the data?
- I want to control a solenoid valve from Digital I/O pins of robot?
- I want to read states of Digital I/O of robot?

Can it be possible from ur_modern_driver? 
I already checked I/O states publish on topic but i need to control the I/O also?

sameergrey, 2017-05-15:
Hi,

I am closing this issue because modern driver already given the I/O states as well as the Service to set the Output pin.
Even analog input pin is giving the states of analog voltage at the output.

closed 2017-05-15 by sameergrey

LuserBen, 2018-07-25:
Hello @sameergrey 
If I want to set the I/O states and the Output pin, which file should I refer to?
Thanks.