ros-industrial-attic/ur_modern_driver #110: Overshoot with position-based ros_control interface caused by internal servoj Pure Pursuit controller?

mortaas, 2017-05-11:
The problems with the position-based controller is already known, and can be corrected to a certain degree by tweaking the parameters: [https://github.com/ThomasTimm/ur_modern_driver/issues/47](url)

My project group tried to use this controller on a UR3-robot, and tried to tweak the parameters to our needs. However we couldn't get this controller to work in a stable manner with the precision we needed. I think I found the root of the problem while documenting our efforts. The parameters defined in the servoj command is usually used in a _Pure Pursuit controller_. These kind of controllers give smooth trajectories, but **doesn't assure that the robot follows the points along the path**. I therefore think that the poor performance is caused by _Pure Pursuit controller_ in the robot not being precise enough for the strict position-based controller in ros_control. We tried loosening the constraints by giving the robot "infinite" time to execute the trajectories, and even though it had a lot of overshoot, it succeeded. In conclusion: the position-based approach in ur_modern_driver should probably be based of movej commands instead.

The picture under _Look Ahead Distance_ in this link illustrates the situation pretty well:[https://se.mathworks.com/help/robotics/ug/pure-pursuit-controller.html](url)

gavanderhoorn, 2017-11-24:
Thanks for your comment @mortaas. Interesting indeed.

> In conclusion: the position-based approach in ur_modern_driver should probably be based of movej commands with blending instead.

At first glance, your statement would seem to go directly against the analysis done by @ThomasTimm and his group ([UR10 Performance Analysis](http://orbit.dtu.dk/en/publications/ur10-performance-analysis(0deaebd1-1593-4641-a971-e1ac0234ed88).html)). Have you read that report?

potiuk, 2018-01-13:
As mentioned in the https://github.com/Zagitta/ur_modern_driver/pull/9 - we seem to have similar problems with overshooting/precision. 

Maybe the reason why it goes against the analysis by @ThomasTimm was that at the time of analysis - with the firmware version available at that time, servoj command used different algorithm - under the hood. UR introduced gain and lookahead_time parameters in servoj in the 3.x series of firmware, which might be a good indication that they changed the algorithm then. I will check with Jacob from UR - we discussed some technical details with him and maybe he cam shed some light on it. 

I also expected much better accuracy when using servoj - much beter than using velocity, and I am a bit disappointed with the results in fact. Since we are right now experimenting with python version of the "Low Bandwith" controler of ours, I might as well try movej + blend - I think the biggest chalelenge with this aproach will be to figure out the right radius for blend - the radius is in tool-space, where we are moving in joint-space and I can expect that for dfiferent poses of robot might require different blend radius - as usual it will be trade-off between accuracy and smoothness of the move. Maybe someone alredy thought about it and has a good/simple proposals of good approach we could use (or some good-enough at least) ?

gavanderhoorn, 2019-11-23:
If this appears to stil be an issue with `ur_robot_driver`, please open an issue on the `ur_robot_driver` issue tracker.

closed 2019-11-23 by gavanderhoorn