ros-industrial-attic/ur_modern_driver #111: UR5 not move with this package and no error output

jacknlliu, 2017-05-13:
I use this package with my universal robot UR5 ( CB3 ).
I run the following commands, and it can connect to robot, and get the `joint states`, but the robot can't move with moveit, and no any feedback from `/execute_trajectory/feedback` and `move_group`. 

```
$ roslaunch ur_modern_driver ur5_bringup.launch limited:=true robot_ip:="192.168.1.125"

$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true
```

The following information are the outputs, but I can't get any error which may cause this. 

Any help will be appreciated!

```
$ roslaunch ur_modern_driver ur5_bringup.launch limited:=true robot_ip:="192.168.1.125"

... logging to /home/ros/.ros/log/a918f77a-37c4-11e7-b73a-0242ac110003/roslaunch-1bcd827b4ebe-385.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /data/UR5/catkin_ws/src/ur_description/urdf/ur5.urdf.xacro
included from: /data/UR5/catkin_ws/src/ur_description/urdf/ur5_joint_limited_robot.urdf.xacro
started roslaunch server http://1bcd827b4ebe:45393/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix:
 * /ur_driver/robot_ip_address: 192.168.1.125
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [399]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a918f77a-37c4-11e7-b73a-0242ac110003
process[rosout-1]: started with pid [412]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [415]
process[ur_driver-3]: started with pid [416]
[ INFO] [1494670437.509568644]: on_goal
Wrong length of message on RT interface: 1078457466
[ INFO] [1494670792.486295224]: on_cancel
[ INFO] [1494670794.786943293]: on_goal
```

---
```
$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

... logging to /home/ros/.ros/log/a918f77a-37c4-11e7-b73a-0242ac110003/roslaunch-1bcd827b4ebe-522.log
Checking log directory for disk usage. This may take awhile.
[... 269 lines cut ...]
    rviz_1bcd827b4ebe_937_1185201707645598711 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[rviz_1bcd827b4ebe_937_1185201707645598711-1]: started with pid [955]
[ INFO] [1494670367.054922200]: rviz version 1.11.15
[ INFO] [1494670367.054983512]: compiled against Qt version 4.8.6
[ INFO] [1494670367.055000940]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1494670367.390986535]: Stereo is NOT SUPPORTED
[ INFO] [1494670367.391078231]: OpenGl version: 3 (GLSL 1.3).
[ INFO] [1494670370.872470198]: Loading robot model 'ur5'...
[ WARN] [1494670370.872550402]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1494670370.872576672]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1494670370.995342952]: Loading robot model 'ur5'...
[ WARN] [1494670370.995434061]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1494670370.995449636]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1494670371.057840158]: Starting scene monitor
[ INFO] [1494670371.063650873]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1494670371.379891501]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1494670372.308197155]: TrajectoryExecution will use old service capability.
[ INFO] [1494670372.308318403]: Ready to take MoveGroup commands for group manipulator.
[ INFO] [1494670372.308422556]: Looking around: no
[ INFO] [1494670372.308480442]: Replanning: no
```

buckmole, 2017-05-17:
YOU CAN TYR THESE 
$ roslaunch ur_modern_driver ur5_ros_control.launch limited:=true robot_ip:="192.168.1.125"

$ roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch limited:=true

$ roslaunch ur5_moveit_config moveit_rviz.launch config:=true

p.s. the file controllers.yaml in ur5_moveit_config/config should be changed as
 "- name:  /vel_based_pos_traj_controller"

jacknlliu, 2017-05-17:
@buckmole  Thank you for your suggestion. I tried it, but the UR Polyscope show the following error, and it can not be removed.
```
Error: speedj() need three parameters (two given)
```

miguelprada, 2017-05-17:
> I use this package with my universal robot UR5 ( CB3 ).
 
Can you give us more details about the specific SW you are running in the CB3? You can find it in the _About_ menu in the Polyscope software.

jacknlliu, 2017-05-17:
@miguelprada  The following is my `PolyScope`  information from its `About` menu,
```
UI: PolyScope 3.1.17336 (Mar 26 2015)
Controller: URControl 3.1.17336 (Mar 26 2015)
Hostname: ur-2015350629
```

miguelprada, 2017-06-05:
It's quite strange, since the [URScript manual](https://s3-eu-west-1.amazonaws.com/ur-support-site/15731/scriptmanual_en.pdf) for version 3.1 clearly states that the third argument to `speedj` is optional.

Have you managed to solve this?

jacknlliu, 2017-06-05:
I think it just seems a warning. What makes me confused is that the robot did not move with moveit. Do you have any ideas to analyze this issue? If you have some idea, please let me know, and I want to have a try.

jacknlliu, 2017-06-25:
I have found the problem. The reverse ip is not corretly configured in `UrDriver::uploadProg()`, since I use a docker container to run the node. I add a parameter for `reverse ip`, now everything works!  I hope making a PR for the newer version of this repository with `reverse ip` setting.

closed 2017-06-25 by jacknlliu

zplizzi, 2017-06-26:
@jacknlliu what did you set the reverse_ip to? I think I'm having the same problem.

zplizzi, 2017-06-26:
Nevermind, figured it out.

jacknlliu, 2017-06-29:
For future readers, more details can be shown here. The `ur_modern_driver` use `local ip address` and `revert port` to make real UR robot know  your host PC, and this can be seen in [UrDriver::uploadProg() ](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp)  line 197 and [bool UrDriver::start()](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp) line 255. 

So you should make your `local ip address`  accessible for UR robot , and `revert port` should keep open on your host. Then the real UR robot connect a TCP server in your `ur_modern_driver` process node which uses your `local ip address` and `revert port` as itself server attribute. 

 But if you run this package inside docker container or VM, [bool UrDriver::start()](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp) line 255 will get a inner ip address of container and VM which is not accessible for external real UR robot, and the 5001 reverse port maybe not accessible from external robot too . 

So you should make the container or VM has an open/external `ip address` and `port`.

cthorey, 2017-07-12:
What change did you make to get it working on the default docker network ? How do you make your container have an open/external ip address and port ?

I somehow cannot get the real robot moving when I'm using the default bridge network to which the container suscribes. Note that everything works fine when the container is on the host network but i'd like to get it working on a bridge network as well. 

Thx for your help

zplizzi, 2017-07-12:
@cthorey I think you can get this working in a docker "bridge" network if you do what @jacknlliu suggested and modify the code to add a "reverse-ip" setting also. Then, when you expose a port in docker (eg `docker run -d -p 80:80 nginx`), and specify the host IP and this exposed port as the reverse address, everything should work - the robot will contact that IP:PORT, and your docker container will respond. 

Without a code modification, I think you'll have to use a host network or perhaps some more fancy docker network setup I'm not familiar with.

cthorey, 2017-07-12:
Thx for the quick answer, I don't have any experience with c++, so I was hoping for a more docker friendly solution. I'll try modifying the code in the meanwhile following your guideline.

jacknlliu, 2017-07-13:
@cthorey I think two things should to be done with the default docker network.
- Add ip address of your host, not the inner ip of docker network

add one line after [ur_driver.cpp Line 255](https://github.com/ThomasTimm/ur_modern_driver/blob/master/src/ur_driver.cpp#L255), set your host ip (for example, your host ip is 192.168.1.123, not the docker ip 172.17.0.1 ), like
```
ip_addr_ = "192.168.1.123"
```
And then rebuild all the project using `catkin` build tools.

- expose needed port when start the docker container
Socket port `50001` should be used, and port `50007` as a redundant design. So `-p 50001:50001` and `-p 50007:50007` can be used with `docker run` command, like
```
$ docker run -p 50001:50001 -p 50007:50007  <YOUR_DOCKER_IMAGE> 
```
Then do other things as usual. Hope this will help you.

cthorey, 2017-07-13:
Thx a lot, it works like a charm :+1: