ros-industrial-attic/ur_modern_driver #112: run into problems of joint speed control, won't rotate given acceleration

Thompson104, 2017-05-18:
I'm trying to publish to /ur_driver/joint_speed for joint speed control. But when I include the acceleration in the "JointTrajectoryPoint", the joint wouldn't rotate at all but it would give you some noise indicating it did receive the information.

`g.points.append(
		JointTrajectoryPoint(
positions=joints_pos_new,velocities=v,accelerations = v_v,time_from_start=rospy.Duration(d_duration)))
`
where `g = JointTrajectory()`

I can easily control the joint without specifying the acceleration part

`g.points.append(
		JointTrajectoryPoint(
positions=joints_pos_new,velocities=v,,time_from_start=rospy.Duration(d_duration)))`
But it doesn''t give me a smooth response

Please note d_duration is updated in each loop. Also, v and v_v here are contant. Another issue is that I don't have this problem when I was using virtualbox on another laptop. 

Any help would be deeply appreciated. Thank you.

Thompson104, 2017-05-18:
Also, there is no error and when I use
` rostopic echo /ur_driver/joint_speed
`
It would give me the right information, but the joints just won't rotate.

shivam7j, 2017-11-14:
@ThomasTimm We are publishing joint velocities using the topics ur_driver/joint_speed and ur_driver/URScript (as mentioned in the GitHub driver) and were able to publish desired joint velocities. However, there was a significant error between the desired joint velocity and actual joint velocity.
And these velocity errors were not linear , they were changing for different velocities , so calibration is also not possible.

Please help on what are the improvement we may need to do to solve this problem.

I am attaching the simple code that we were publishing
[vel_test.zip](https://github.com/ThomasTimm/ur_modern_driver/files/1470487/vel_test.zip)

gavanderhoorn, 2019-06-29:
Closing as this was reported against `master` branch. `kinetic-devel` changed things significantly.

If this is still an issue and someone runs into it, we can re-open.

closed 2019-06-29 by gavanderhoorn