ros-industrial-attic/ur_modern_driver #113: UR10 Changed Joint_names?

ipa-mjp, 2017-05-23:
Hello,
I am trying to use UR10 modern driver. It changes Joint_names according to alphabetic order.

Originally:
name:['arm_shoulder_pan_joint', 'arm_shoulder_lift_joint', 'arm_elbow_joint', 'arm_wrist_1_joint', 'arm_wrist_2_joint', 'arm_wrist_3_joint']

After reordering:
name: ['arm_elbow_joint', 'arm_shoulder_lift_joint', 'arm_shoulder_pan_joint', 'arm_wrist_1_joint', 'arm_wrist_2_joint', 'arm_wrist_3_joint']

How can it be solved?
In program, where are you reordering it?

Any help would be deeply appreciated. Thank you.

miguelprada, 2017-06-05:
If you're using the `ros_control` enabled version of the driver, the reordering is performed internally by the `InterfaceManager` when the hardware interfaces (i.e. joints) are [stored](https://github.com/ros-controls/ros_control/blob/a632517ab7b560196ae835651af7c5e9b4c51ede/hardware_interface/include/hardware_interface/internal/interface_manager.h#L64) inside a [std::map](https://github.com/ros-controls/ros_control/blob/a632517ab7b560196ae835651af7c5e9b4c51ede/hardware_interface/include/hardware_interface/internal/interface_manager.h#L96).

I don't think you can easily modify this.

ipa-mjp, 2017-06-05:
Thank a lot.
I found the easiest solution is to renaming the joint_names.

closed 2017-07-21 by ThomasTimm