ros-industrial-attic/ur_modern_driver #114: VelocityJointInterface gets UR10 unstable

ane-fernandez, 2017-05-25:
As soon as I start the robot using a velocity controller, the robot gets unstable, even without sending any movement. I have checked that depending the value of P gain parameter, the time needed to gets unstable is different: the highest it is, the sooner gets unstable.

I have read the #55 and #92 issues and I have double checked that I have an updated driver version, calling to speedj properly with t=0.008. However the robot behavior is not good. 

To try to understand what was going on, I have monitored the velocity using @miguelprada ur_hardware_interface.cpp modified version and I got this (P value for all joints is 1.0) :

[image]

The blue line is the output of the controller, the red line is the target joint velocity as reported by the UR and the yellow line is the actual joint velocity. 

What I observe is, as in #55 issue, that there is a delay between velocity controller commands and the target joint velocity reported by the UR.  There is also a delay between the target velocity and joint current velocity. Another issue that I'm worried about are the jumps that appears on the joint trajectory controller out. 

I have read all the issues related with this topic, but I'm not able to solve the problem. s someone having the same problems? Any idea how to fix it? The firmware version is: 3.3.4.310

Thanks in advance

miguelprada, 2017-05-29:
@ane-fernandez, which control box version do you have?

We've been recently having similar issues with some of our robots and according to our ongoing tests, it only happens with CB3 (not with CB3.1), even when the software versions match.

ane-fernandez, 2017-05-29:
Thank you very much @miguelprada, you confirm my suspicions. In our laboratory we have an CB2 and CB3. I have been doing several test with both CB (using the same external code in both cases ) and the result is that with CB2 everything works well and with CB3 I have the behaviour described in the post before.

I've been running the ur_modern_driver with different t parameters and the behaviour of the robot changes. I would like to make more tests, but what I observes is:

- With a value grater than 0.008 the robot don't get unstable but the #55 problem appear
- With a lower value, the robot don't get unstable but the velocity commanded is not achieve.

mitch1044444, 2017-06-28:
@ane-fernandez 
Hey, I wonder if you could share your code or suggestions on how to plot this? I'm relatively new to Ur robot, and I'm trying to learn UR script and other ways to control this robot. I would really appreciate your help.

ane-fernandez, 2017-06-29:
Hi @mitch1044444  I don't really have any code. I used  [ur_hardware_interface.cpp](https://github.com/tecnalia-advancedmanufacturing-robotics/ur_modern_driver/blob/34654a07140a95dc36c2b415b33792fde366d6e6/src/ur_hardware_interface.cpp) that @miguelprada mentioned in the #55 issue. This hardware interface publishes in the joint_state topic, the current velocity, the output of the controller and the UR current target velocity of each joint. I save them in a file and then plot them using office. That is the way I get this graphic

htugal, 2017-10-24:
Hi @ane-fernandez  could you able to fix the problem particularly the jumps that appear in the velocity signal ?

herczegzoltan, 2017-11-20:
Hello,

I am working on a project, I have to use velocity controller. Could you help me out? How can I change the default controller to velocity or position? I have been working on this for a while without success 
Thank you.

gavanderhoorn, 2017-11-24:
@herczegzoltan: please don't hijack issues like this. See whether #149 clarifies things for you. If not, please open *a new issue*. Also: please try to make sure you absolutely *need* to use `ros_control`. If not, perhaps the `joint_speed` topic would be enough.

Olimoyo, 2018-09-19:
@ane-fernandez: Were you able to solve this problem or find out what was causing it?

closed 2019-11-23 by gavanderhoorn