ros-industrial-attic/ur_modern_driver #118: UR3 robot does not move

Djokovalle, 2017-07-11:
Hello,

I am trying to get the UR3 working with the ur_modern_driver, but every time I try to move the robot, with RViz and with moveit_commander, the robot stays still and there is no error message, it just says on_goal and waits. 
But my configuration and code works fine for the ur3 simulation with gazebo and the robot follows the planned paths in gazebo.
For the simulation I only replace the bringup.launch with:
`roslaunch ur_gazebo ur3.launch`

[roslaunch ur_gazebo ur3.launch.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138673/roslaunch.ur_gazebo.ur3.launch.pdf)

For the real robot:

`roslaunch ur_modern_driver ur3_bringup_joint_limited.launch robot_ip:=ROBOT_IP`
`roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch`
`roslaunch ur3_moveit_config moveit_rviz.launch config:=true`

The outputs in the terminal are the following:
[ur3_bringup_joint_limited.launch.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138466/ur3_bringup_joint_limited.launch.pdf)

[ur3_moveit_config ur3_moveit_planning_execution.launch .pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138470/ur3_moveit_config.ur3_moveit_planning_execution.launch.pdf)

[ur3_moveit_config moveit_rviz.launch.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138479/ur3_moveit_config.moveit_rviz.launch.pdf)

I added the rqt_graph:
[image]

These are my launch files:
[ur3_bringup.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138687/ur3_bringup.pdf)
[ur3_moveit_planning_execution.launch.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138688/ur3_moveit_planning_execution.launch.pdf)
[moveit_rviz.launch.pdf](https://github.com/ThomasTimm/ur_modern_driver/files/1138689/moveit_rviz.launch.pdf)
Thank you in advance for your help!!!

Djokovalle, 2017-07-20:
Thank you @cthorey for the referenced issue 119/111/118.
Unfortunately, this does not apply to my situation, because I am not using Docker nor a VM. Even with a direct connection in from of an ethernet cable and a localhost setup, the problem stays the same.
Any other tips?

ThomasTimm, 2017-07-21:
Have you tried to see if you can move the robot with the test_move.py script?

Djokovalle, 2017-07-21:
Hello Thomas,

yes I am currently trying to move the robot with the test_move.py. In addition, I downloaded the modified version of your modern driver from Zagitta.
Unfortunately there is still no movement of the robot.
When we launch the ur3_bringup file and enable the robot by "rosservice call /ur_driver/robot_enable" the execution of the test_move.py leads to the output:

valentin@valentin-System-Product-Name:~$ roslaunch ur_modern_driver ur3_bringup.launch robot_ip:=156.18.37.223
... logging to /home/valentin/.ros/log/9cedb906-6e0e-11e7-97e5-bcaec5300cd6/roslaunch-valentin-System-Product-Name-28486.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

redefining global property: pi
when processing file: /home/valentin/catkin_new/src/universal_robot/ur_description/urdf/ur3.urdf.xacro
included from: /home/valentin/catkin_new/src/universal_robot/ur_description/urdf/ur3_robot.urdf.xacro
started roslaunch server http://valentin-System-Product-Name:45853/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 3.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix:
 * /ur_driver/robot_ip_address: 156.18.37.223
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [28500]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9cedb906-6e0e-11e7-97e5-bcaec5300cd6
process[rosout-1]: started with pid [28513]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [28526]
process[ur_driver-3]: started with pid [28531]
[ INFO] [1500639471.522380744]: Setting up connection: 156.18.37.223:30003
[ INFO] [1500639471.522622865]: Connection established for 156.18.37.223:30003
[ INFO] [1500639471.522667622]: Setting up connection: 156.18.37.223:30001
[ INFO] [1500639471.522846798]: Connection established for 156.18.37.223:30001
[ INFO] [1500639471.523029479]: Got VersionMessage:
[ INFO] [1500639471.523050836]: project name: URControl
[ INFO] [1500639471.523063240]: version: 3.4.0
[ INFO] [1500639471.523075390]: build date: Mar 08 2017, 15:04:37
[ INFO] [1500639471.523088265]: Disconnecting from 156.18.37.223:30001
[ INFO] [1500639471.537150493]: Setting up connection: :50001
[ INFO] [1500639471.537326262]: Connection established for :50001
[ INFO] [1500639471.537370717]: ActionServer enabled
[ INFO] [1500639471.539408165]: Starting main loop
[ INFO] [1500639471.539800964]: Setting up connection: 156.18.37.223:30003
[ INFO] [1500639471.540076843]: Connection established for 156.18.37.223:30003
[ INFO] [1500639471.540161157]: Setting up connection: 156.18.37.223:30002
[ INFO] [1500639471.540390250]: Connection established for 156.18.37.223:30002
[ INFO] [1500639471.542232874]: Starting ActionServer
[ INFO] [1500639471.542327100]: Trajectory thread started
[ INFO] [1500639526.738287884]: Received new goal
[ INFO] [1500639526.738467534]: Trajectory received and accepted
[ INFO] [1500639526.738552035]: Translating trajectory
[ INFO] [1500639526.738626305]: Executing trajectory with 31 points and duration of 40.000s
[ INFO] [1500639526.738672528]: Uploading trajectory program to robot

At this point the test_move waits and the robot doesn't move.

I appreciate your mail and thank you in advance for any further help.

Valentin

Djokovalle, 2017-07-21:
For further information, here is the output of the follow_joint_trajectory/goal after I launch move_test.py. We only change the Q1, Q2, Q3 to have very small movement close to our initial position:

valentin@valentin-System-Product-Name:~$ rostopic echo /follow_joint_trajectory/goal
header: 
  seq: 1
  stamp: 
    secs: 1500640833
    nsecs: 144717931
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1500640833
    nsecs: 144656896
  id: /test_move_29224_1500640830024-1-1500640833.145
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: ''
    joint_names: ['shoulder_pan_joint', 'shoulder_lift_joint', 'elbow_joint', 'wrist_1_joint', 'wrist_2_joint', 'wrist_3_joint']
    points: 
      - 
        positions: [0.840342104434967, -0.4949100653277796, -2.4966488520251673, -1.142003361378805, 1.572191834449768, -0.0430830160724085]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0

...

...

        positions: [0.872665, -0.401426, -2.44346, -1.09956, 1.51844, 6.26573]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: []
        effort: []
        time_from_start: 
          secs: 40
          nsecs:         0
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0

maxime-petit, 2017-09-04:
Hi all,

I am working with @Djokovalle, and continue his work with the UR3. 

Do you have any inputs about what the problem could be? Is it some wrong launching file/ROS?

Might it be because of the Polyscope version (3.4.0.39), from March 2017?

Thanks in advance for the support

gavanderhoorn, 2017-09-04:
At this point I have no idea what could be the cause, but first thing would be to check the robot's log. Try running `test_move.py` and keep an eye on the log. Log into the controller and add the relevant lines here as a comment.

maxime-petit, 2017-09-20:
Hi,

Sorry for the delay.

Launching the ur modern driver produce the following

```
SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 3.0
 * /ur_driver/max_velocity: 0.1
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 156.18.37.233
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [18977]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4ef02796-9e1c-11e7-8e75-bcaec5300cd6
process[rosout-1]: started with pid [18990]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [19003]
process[ur_driver-3]: started with pid [19008]
[ INFO] [1505923009.440916447]: Setting up connection: 156.18.37.233:30003
[ INFO] [1505923010.446022237]: Connection established for 156.18.37.233:30003
[ INFO] [1505923010.446105602]: Setting up connection: 156.18.37.233:30001
[ INFO] [1505923010.446340191]: Connection established for 156.18.37.233:30001
[ INFO] [1505923010.446602848]: Got VersionMessage:
[ INFO] [1505923010.446630877]: project name: URControl
[ INFO] [1505923010.446649166]: version: 3.4.0
[ INFO] [1505923010.446666374]: build date: Mar 08 2017, 15:04:37
[ INFO] [1505923010.446689238]: Disconnecting from 156.18.37.233:30001
[ INFO] [1505923010.464637292]: Setting up connection: :50001
[ INFO] [1505923010.464784120]: Connection established for :50001
[ INFO] [1505923010.464820062]: ActionServer enabled
====> max_velocity= 2.11242e-317
[ INFO] [1505923010.470061164]: Starting main loop
[ INFO] [1505923010.470190528]: Setting up connection: 156.18.37.233:30003
[ INFO] [1505923010.470366053]: Setting up connection: 156.18.37.233:30002
[ INFO] [1505923010.470477157]: Connection established for 156.18.37.233:30003
[ INFO] [1505923010.470581981]: Connection established for 156.18.37.233:30002
[ INFO] [1505923010.471890722]: Starting ActionServer
[ INFO] [1505923010.471980621]: Trajectory thread started
```

Then launching the `test_move.py` add the following:
```
[ INFO] [1505923061.718124142]: Received new goal
[ WARN] [1505923061.718191805]: Goal error: Robot is not ready, check robot_mode
```

Probably linked to https://github.com/ros-industrial/universal_robot/issues/191 ?

BTW If I run the moveit with rviz, I have the actual states of the robot.

gavanderhoorn, 2017-09-20:
Two things look weird to me:

> `====> max_velocity= 2.11242e-317`

and

> `/ur_driver/max_velocity: 0.1`

Neither value is what I would expect. Defaults are (for all variants) `10.0`. Did you change anything there?

As to:

> ```
> [ INFO] [1505923061.718124142]: Received new goal
>[ WARN] [1505923061.718191805]: Goal error: Robot is not ready, check robot_mode
> ```

That is not good. I'll have to check whether your exact Polyscope version is already supported by #120.

maxime-petit, 2017-09-27:
Hi,

Thanks for your reply @gavanderhoorn 

I will try to pull and reinstall a clean and up-to-date version. 

However, I am not sure if I should pull the current github (https://github.com/ThomasTimm/ur_modern_driver) or the fork from @Zagitta (https://github.com/Zagitta/ur_modern_driver) 

If the Polyscope version is not supported, should I try to downgrade our robot? 

According to https://github.com/ThomasTimm/ur_modern_driver/issues/80, CB3 (3.3.1.237) should be fine right?

gavanderhoorn, 2017-09-27:
At this point the rewrite in #120 is probably best to test.

> If the Polyscope version is not supported, should I try to downgrade our robot?

that, or figure out what changed that made it not work. Then ideally submit a PR against #120 to get that fixed.

> According to #80, CB3 (3.3.1.237) should be fine right?

may be, but in the output that you post I see:

> ```
> [ INFO] [1505923010.446602848]: Got VersionMessage:
> [ INFO] [1505923010.446630877]: project name: URControl
> [ INFO] [1505923010.446649166]: version: 3.4.0
> ```

`3.4.0` is not `3.3.1.x`. And the PR description on #120 mentions that it was tested with versions from `1.8.x` till `3.3.4`.

It could well be that the cause of your problems is something completely different, but it is something to check.

gavanderhoorn, 2017-11-24:
Is this still an issue @maxime-petit?

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

@Djokovalle: if this is still a problem for you, please re-open and provide more information.

closed 2018-03-02 by gavanderhoorn

brianjohnzhang, 2019-03-10:
@Djokovalle @maxime-petit I had the same issue; when looking at the logs on the Polyscope, I found that it silently needs port 50001 to be open on your computer for a connection. Not sure if this is detectable/placeable in a check somewhere, but after opening it everything works. Thanks gavanderhoorn for the reminder to check the logs!