ros-industrial-attic/ur_modern_driver #121: test_move.py   => waiting on server...

htugal, 2017-07-18:
I know this issue has been already mentioned some other places, but i've tried every solution suggested and i still cant get it to work

Im following the tutorials for the Universal_robot package. I've established a connection
recieved:

[2017-07-18 15:25:40.832867] Out: hello
[INFO] [WallTime: 1500387940.839168] Robot connected

but the test_move.py is stuck on waiting for server...

My ROS is 

 * /rosdistro: indigo
 * /rosversion: 1.11.21

Thanks,

closed 2017-07-19 by htugal

vkguda355, 2019-07-15:
how did u resolve this issue, i am also having the same problem.

htugal, 2019-07-16:
Hi, 

I don't recall how sorted out that issue, yet  try to ping (ping 'ip') the robot ip to see whether  there is any issue with the connection or not. Then, echo the topics (joint or Cartesian positions for instance) being published by the online robot. If everything is OK by now, you can publish joint velocities to the robot 
ros::Publisher arm_pub = nh.advertise<trajectory_msgs::JointTrajectory>("ur_driver/joint_speed",10);
the robot will move accordingly. 
I hope that helps,