ros-industrial-attic/ur_modern_driver #122: UR5 executes partial trajectory

atenpas, 2017-07-19:
When using the **joint trajectory action** (JTA), our UR5 robot sometimes executes only a partial trajectory. There's no error message shown on the teaching pendant, and the "result" from the ROS action is "True". 

To be more precise, I can usually use the JTA a few times (2-3), executing the complete trajectory **without** problems, and then for a new goal, it only does a few steps and stops moving. The trajectories  send to the JTA consist of two points (current c-space config, target c-space config).

I've attached the output on the two ROS topics _follow_joint_trajectory/goal_ and _follow_joint_trajectory/result_.

Some more info:
- **Polyscope version:** 3.1.18024
- **Firmware version:** 3.1.18024
- **ROS Indigo, Ubuntu 14.04**

[fjt_goal.txt](https://github.com/ThomasTimm/ur_modern_driver/files/1160558/fjt_goal.txt)
[fjt_result.txt](https://github.com/ThomasTimm/ur_modern_driver/files/1160559/fjt_result.txt)

gavanderhoorn, 2017-11-24:
@atenpas: I only see msgs on the `result` topic with this:

```yaml
result: 
  error_code: 0
  error_string: Robot was halted
```

That should only be returned in case the driver believes the robot has been emergency or protective stopped:

https://github.com/ThomasTimm/ur_modern_driver/blob/af6edc8a3552c449d27076db1642adf8afa97714/src/ur_ros_wrapper.cpp#L780-L787

I'm assuming that wasn't the case?

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

@atenpas: if this is still a problem for you, please re-open and provide more information.

closed 2018-03-02 by gavanderhoorn