ros-industrial-attic/ur_modern_driver #123: UR10 with CB3 and ROS

alexsavio, 2017-07-21:
Hi,

I have to work with an UR10 with the latest version of CB3.
I would like to build an application to steer and monitor in real time the robot in Ubuntu.
Does this driver work without ROS?

For me python_urx fits quite well to my problem, however the client is worried about latency because of Python. Do you have any reference/measures about this?

Thank you very much!

gavanderhoorn, 2017-07-21:
I think the best references for this (at least in the context of this driver) would be [Optimizing the Universal Robots ROS driver](http://www.forskningsdatabasen.dk/en/catalog/2287861023) and [UR10 Performance Analysis](http://orbit.dtu.dk/en/publications/ur10-performance-analysis(0deaebd1-1593-4641-a971-e1ac0234ed88).html).

> the client is worried about latency because of Python

really ..

ThomasTimm, 2017-07-21:
Yes, you can easily use the driver without ROS, just replace the ur_ros_wrapper.cpp with your own wrapper. Also, you dont need the file ur_hardware_interface.cpp which is for ros_control. None of the other files depends on ROS.

But, as hinted above, python isn't going to introduce any bottlenecks in the control of the robot.

closed 2017-07-21 by ThomasTimm