ros-industrial-attic/ur_modern_driver #124: socket_read binary_integer: timeou

nag92, 2017-07-24:
I am trying to control a UR10 arm with moveit and the ur_modern_driver. If moveit get a bad trajectory I get the following error on the UR10 tablet under the log: socket_read binary_integer: timeout
After that error appears I am no longer able to control the arm. I have to relaunch the node that is controlling the arm. Is this a known issue, I cannot seem to fix it, not figure out the cause?

gavanderhoorn, 2017-11-24:
> If moveit get a bad trajectory [..]

What is a "bad trajectory"?

gavanderhoorn, 2018-03-02:
Closing due to inactivity.

@nag92: if this is still a problem for you, please re-open and provide more information.

closed 2018-03-02 by gavanderhoorn

paulyang1990, 2018-11-14:
please reopen this issue. I encountered this problem as well. 

My case is: I used Moveit to plan a trajectory and send it to the robot. The robot does not stop when it reaches goal state but continues to move. At the same time, the log of the robot shows "socket_read binary_integer: timeout"